eclipse-upm/upm #34: UPM drive for SSD1351 OLED display

stanwgifford, 2015-01-06:
This is a placeholder for a upm module to control the SSD1351 oled display. This is a SPI device and the module I will be using when writing the code is a freetronics module ( http://freetronics.com/oled128 )

stanwgifford, 2015-02-11:
Note - I am probably going to put this on hold until a new kernel comes that gives a decent SPI data rate - currently using the arduino code, the baud rate across SPI to the OLED looks like it is running at 300 baud (well pretty slowly). Note - the standard libraries for this OLED do work well in the Arduino environment - just slow.............

sukeshak, 2015-08-11:
The Oled used with Sparkfun Edison block also uses SPI and it works great without any lag. 
I am planning to use a 1.5" 128x64 (blue) instead of that... You think I should also wait for the new kernel?

sukeshak, 2015-10-10:
Any updates on this? Looking forward for this driver :)

sukeshak, 2015-11-25:
@stanwgifford  any update on this?

Propanu, 2016-03-05:
Driver was made available today with basic functionality implemented (lightweight version of Adafruit's GFX lib)!

closed 2016-03-05 by Propanu

sukeshak, 2016-03-06:
Thank you for getting this done... Sounds great... will test it and revert in a day