eclipse-upm/upm #99: I don't understand...

TinajaLabs, 2015-01-29:
I get it that this is code that is supposed to work with edison and the grove line of sensors and actuators.

Is there a tutorial that describes how you would put them all together?  How would I put together a house full of sensors and track, for example, temperature in every room of my house?

Thanks,
Chris.

jrvandr, 2015-01-29:
Correct, this is the github repository for the sensors supported by Galileo and Edison.
Some resources which may be helpful for the project you're interested in are:
	Main Intel Maker page:	http://maker.intel.com
	Forums & Support:	http://www.intel.com/content/www/us/en/do-it-yourself/forums-and-support.html
	Project Gallery:	http://www.intel.com/content/www/us/en/do-it-yourself/project-gallery.html

Hope that helps get you started on your journey with this project.
Best of luck!

Propanu, 2015-01-29:
For a home automation project like the one you're describing, it might be useful to look into some cloud based options as well:

https://software.intel.com/en-us/intel-iot-developer-kit-cloud-based-analytics-user-guide

closed 2015-02-25 by jrvandr