eclipse-upm/upm #112: LoL actuator library failure

michaesc, 2015-02-17:
PROBLEM

When running github:///upm/master/examples/lol-example.cxx the Olimex LoL lights up but in a rather uncontrolled way. Even doing something simple like:

int main(int argc, char **argv)
{
    sensor = new upm::LoL();
    sensor->setPixel(1, 1, 1);
    sensor->setPixel(1, 2, 1);
    sensor->setPixel(1, 3, 1);
    sensor->setPixel(1, 4, 1);
    usleep(100000000);
}

...causes pixels to flash and turn off almost immediately and then repeat (even though there is no loop.) Some of the pixel flashes are in a pyramid form (matching the charliplexing algorithm.)

CONFIGURATION

I'm using a Galileo Gen1 with up to date firmware 1.0.4 on SPI and the latest IoT Devkit 1.1 on a 8 Go SDHC. I'm booting to Devkit of course.

# uname -a
Linux hostname 3.8.7-yocto-standard #1 Wed Sep 3 10:41:56 BST 2014 i586 GNU/Linux

#cat /etc/opkg/mraa-upm.conf
src mraa-upm http://iotdk.intel.com/repos/1.1/intelgalactic

# opkg list | grep -i lol
# opkg list | grep -i upm
upm - 0.1.9 - upm built using CMake
Source: https://github.com/intel-iot-devkit/upm
Description: upm built using CMake
Installed-Time: 1424137558

# find /lib/ -name '*lol*'
# find /lib32/ -name '*lol*'
# find /usr/lib/ -name '*lol*'
/usr/lib/node_modules/jsupm_lol
/usr/lib/node_modules/jsupm_lol/jsupm_lol.node
/usr/lib/libupm-lol.so
/usr/lib/libupm-lol.so.0
/usr/lib/python2.7/site-packages/pyupm_lol.py
/usr/lib/python2.7/site-packages/_pyupm_lol.so
/usr/lib/libupm-lol.so.0.1.9
/usr/lib/pkgconfig/upm-lol.pc

QUESTIONS

What do I need to change to properly control the LoL shield? Build a new libupm-lol from source? Which sources are working?

Propanu, 2015-02-18:
See issue #11. The problem is that the GPIO pins on the board cannot be switched fast enough as required by the Charlieplexing algorithm to work properly. The LoL Shield has 14 pins to drive the LEDs so if you want a refresh rate of 50Hz you'll need to write the pins at 14 x 50 = 700 Hz. The GPIO pins will do about 230 Hz as listed in the Galileo F.A.Q. (https://communities.intel.com/docs/DOC-22543 under Building).

Won't fix as this is a hardware issue, besides, the LoL Shield has been discontinued.

michaesc, 2015-02-18:
What about Galileo Gen2 boards, and is it true that libupm-lol works as it should on them? I heard that libupm-lol on Gen2 causes the GPIO kernel driver to deadlock because of an incorrect locking implementation, so my next question is 'where is the source code for this incorrect locking implementation?' Is it in libupm-lol, libupm, libmraa, the Linux kernel, or some other component?

...or is libupm-lol working correctly on Galileo Gen2 boards?

Propanu, 2015-02-21:
It would be hard to pinpoint the exact reason causing the kernel freeze, and from my understanding it can happen quite often when bit banging the GPIO pins. Even external interference can trigger this thus it's not necessarily a specific line of code. Theoretically though, UPM shouldn't have to lock anything.

To answer your other question, the shield will not work correctly on any arduino board that doesn't meet the speed requirement I listed above.

closed 2015-02-25 by jrvandr