eclipse-upm/upm #114: Buzzer PWM state indeterminate after use

agnathan, 2015-02-19:
Several people at the IoT Roadshows have run into an issue with the Buzzer remaining on after the playSound function delay has ended.

Let's say that a call is made to playSound(chord[chord_ind], 1000000).  It appears that the buzzer's PWM signal will continue for the 1000000 usecs, but when it ends the state is either high or low.  If it's high then the buzzer never turns off.

Our work around is to declare another gpio pin on and write low to it after playing a sound.

Here's the code that has the bug. It's just the buzzer sample.

#include <unistd.h>
#include <iostream>
#include <signal.h>
#include "buzzer.h"

int
main(int argc, char **argv) {
    //! [Interesting]
    int chord[] = { DO, RE, MI, FA, SOL, LA, SI, DO, SI };

    // create Buzzer instance
    upm::Buzzer* sound = new upm::Buzzer(5);
    // print sensor name
    std::cout << sound->name() << std::endl;

    // play sound (DO, RE, MI, etc...), pausing for 0.1 seconds between notes
    for (int chord_ind = 0; chord_ind < 7; chord_ind++) {
        // play each note for one second
        std::cout << sound->playSound(chord[chord_ind], 1000000) << std::endl;
        usleep(100000);
    }
    //! [Interesting]

    std::cout << "exiting application" << std::endl;

    delete sound;

    return 0;
}

Propanu, 2015-03-02:
Do you have an example for your solution? Is it this (pwm_pin is the same pin the PWM was initialized on):

mraa::Gpio* d_pin = new mraa::Gpio(pwm_pin);
d_pin->dir(mraa::DIR_OUT);
d_pin->write(0);

This doesn't seem to work all the time and even if it does, there's the multiplexing overhead. I did however find a different fix for this issue.

Propanu, 2015-03-09:
Fixed in 0aa3d594877172f6e8c9860332c04da72b3e8820

closed 2015-03-09 by Propanu

eldrago, 2015-08-12:
This might fix the buzzer issue if you are coding in C, however I am still running into this issue when I am using node.js within the XDK.

Propanu, 2015-08-13:
The fix was tested for all the supported languages. As a matter of fact I developed it with node first using MRAA and only after ported it to UPM. Please make sure you are using the latest image for the Intel Edison along with the latest libraries. The Buzzer API also provides a function called stopSound() that will force the PWM on the pin to 0.

eldrago, 2015-08-13:
I must be missing something then. Maybe I only updated the c libraries or something.    I am over in JF 5, is there any chance i could take a few minutes of your time? 

<div>-------- Original message --------</div><div>From: Propanu <notifications@github.com> </div><div>Date:08/13/2015  11:16 AM  (GMT-08:00) </div><div>To: intel-iot-devkit/upm <upm@noreply.github.com> </div><div>Cc: eldrago <eldrago2@yahoo.com> </div><div>Subject: Re: [upm] Buzzer PWM state indeterminate after use  (#114) </div><div>
</div>The fix was tested for all the supported languages. As a matter of fact I developed it with node first using MRAA and only after ported it to UPM. Please make sure you are using the latest image for the Intel Edison along with the latest libraries. The Buzzer API also provides a function called stopSound() that will force the PWM on the pin to 0.

Propanu, 2015-08-13:
I’m available this afternoon. Send me a meeting invite.

Tudor

eldrago, 2015-08-13:
Yeah I verified it was my bad, Thank you for the help Propanu!

eldrago, 2015-08-14:
Heya Tudor,
Just tried out the test code with the updated upm and mraa.  It works MUCH better, thanks again for your help.
However, I did notice one thing.  When it finishes playing a sound using the timeout the sound doesn't completely go away, it just plays at a very low volume.  Looking at the scope it looks like the pwm keeps going, just with a minuscule duty cycle, this is easily worked around by throwing in a stopSound command.  Which works for me, I just thought I would let you know.
Thanks again,Zack