eclipse-upm/upm #139: Hx711 ADC support

rafaneri, 2015-03-13:
Hello I have this [ADC](http://www.dfrobot.com/image/data/SEN0160/hx711_english.pdf) and I was wondering if there plan for support it yet? I had some problems to use it with node.js. I tried do a addon, but the result was the same that using node.js with mraa lib. Can you help me with this? This is the [repository](https://github.com/rafaneri/hx711-addon).

Thank you!

jrvandr, 2015-03-13:
Hi, thank you for the inquiry!  Currently we don't have support for this sensor.  We are always looking to add new sensors, so we can take a look to see when/if it can be added.  We will take a look at the spec sheet and sensor code from the link you posted and will be happy to provide any assistance we can.
Thank you!

rafaneri, 2015-03-15:
Hi @jrvandr, I did changes to leave the code as the data sheet, then it return the raw ADC data. Now it's working!
Thank you!

jrvandr, 2015-03-18:
Great to hear it is working for you!  Submitting a pull request will allow us to get this sensor added to the repository.
Thank you.

rafaneri, 2015-03-21:
Hello @jrvandr, can you help me with a doubt?
I did a fork on upm, wrote my module and install on galileo, but when try execute the example I got the error on attached image.

[Repository](https://github.com/rafaneri/upm)

[image]

rafaneri, 2015-03-21:
Ok! It's a wrong parameter -upm-hx711, it should be -lupm-hx711! I'm going do a pull request!

closed 2015-03-26 by Propanu