eclipse-upm/upm #169: bad arguments for node/python getSampledWindow grove mic samples

hh, 2015-04-20:
In trying the samples on a fresh load of w05-15 and connecting the sound sensor, I get the following errors:

```
root@eddiebetter:~# node mic.js

/home/root/mic.js:46
    var len = myMic.getSampledWindow(2, 128, buffer);
                    ^
Error: in method 'Microphone_getSampledWindow', argument 4 of type 'uint16_t *'
    at Object.<anonymous> (/home/root/mic.js:46:21)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
root@eddiebetter:~# python mic.py 
Traceback (most recent call last):
  File "mic.py", line 39, in <module>
    len = myMic.getSampledWindow(2, 128, buffer);
  File "/usr/lib/python2.7/site-packages/pyupm_mic.py", line 152, in getSampledWindow
    return _pyupm_mic.Microphone_getSampledWindow(self, *args)
TypeError: in method 'Microphone_getSampledWindow', argument 4 of type 'uint16_t *'
```

Zion-ICS, 2015-04-20:
Thanks for the bug report!
Here's my first question: which version of [SWIG](http://swig.org/) do you have installed?

hh, 2015-04-20:
I have a fresh load of:
http://downloadmirror.intel.com/24698/eng/edison-image-ww05-15.zip
Onto an edison w/ arduino break out board.
I ran other examples (rgblcd/speaker) and the worked fine.
Is there something special I need to do to copy those  mic libraries into place? I thought the shipping xdk included them?

Propanu, 2015-04-21:
You are using an old version of MRAA/UPM that doesn't our node typemap updates. With your Edison connected to the internet, try the steps described here to update your libraries:
https://software.intel.com/en-us/node/532592

Let us know if this fixes your problem, our code samples were tested and working fine with upm 0.1.9.

closed 2015-05-07 by Propanu