eclipse-upm/upm #170: grove rotary function mismatch

hh, 2015-04-20:
It may be that the version of upm that shipped with w05-15 is different than master here.
```
root@eddiebetter:~# node groverotary.js 

/home/root/groverotary.js:29
var groveRotary = new upm_grove.GroveRotary(0);
                  ^
TypeError: undefined is not a function
    at Object.<anonymous> (/home/root/groverotary.js:29:19)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
root@eddiebetter:~# python groverotary.py
Traceback (most recent call last):
  File "groverotary.py", line 27, in <module>
    knob = grove.GroveRotary(0)
AttributeError: 'module' object has no attribute 'GroveRotary'

```

hh, 2015-04-20:
This is being run on http://downloadmirror.intel.com/24698/eng/edison-image-ww05-15.zip

mceisel, 2015-04-21:
The problem is due to an old version of  mraa/upm. From the edison CLI, connect to the internet using "configure_edison --wifi",  then "opkg update", and "opkg upgrade".

You can see the node method availables as follows:

//setup/Initialization
var upm_grove = require('jsupm_grove');

console.log('CLASS');
for (var m in upm_grove) {
   console.log (typeof upm_grove[m] + ': ' + m);
}

//setup access analog input Analog pin #0 (A0)
var groveRotary = new upm_grove.GroveRotary(0);

console.log('INSTANCE');
for (var m in groveRotary) {
   console.log (typeof groveRotary[m] + ': ' + m);
}

closed 2015-05-07 by Propanu