eclipse-upm/upm #172: SSD1327: Segfault when calling the draw method with incorrect arguments from python

vvvverre, 2015-04-20:
When calling ssd1327.draw() from python the library segfaults if the first argument is not of type lcd.uint8Array. It would be much nicer if the library returns an error when it finds an unexpected type, instead of carrying on with a null pointer (see data in the backtrace below) and causing a guaranteed segfault down the line. Ideally passing native python types would work, but I think this would require the use of SWIG typemaps?

The following is a reproducer:

>Python 2.7.3 (default, Mar 31 2015, 11:37:49) 
>[GCC 4.8.2] on linux2
>Type "help", "copyright", "credits" or "license" for more information.
> \>\>\> import pyupm_i2clcd as lcd
> \>\>\> x = lcd.SSD1327(0);
> \>\>\> x.draw("\x55", 1); 
>Program received signal SIGSEGV, Segmentation fault.
>0xb7efe7b3 in upm::SSD1327::draw (this=0x805be00, data=0x0, bytes=1) at >/home/root/git/upm/src/lcd/ssd1327.cxx:146
>146	            uint8_t bitOne = (data[row] << col) & 0x80;

A backtrace from gdb is shown below:

>(gdb) bt
>#0  0xb7efe7b3 in upm::SSD1327::draw (this=0x805be00, data=0x0, bytes=1) at /home/root/git/upm/src/lcd/ssd1327.cxx:146
>\#1  0xb7ef9573 in _wrap_SSD1327_draw (args=0xb7f4d8ec) at /home/root/git/upm/build/src/lcd/pyupm_i2clcdPYTHON_wrap.cxx:5325
>\#2  0x4a163aaf in PyCFunction_Call () from /usr/lib/libpython2.7.so.1.0
>\#3  0x4a1c3dd4 in PyEval_EvalFrameEx () from /usr/lib/libpython2.7.so.1.0
>\#4  0x4a1c3a7d in PyEval_EvalFrameEx () from /usr/lib/libpython2.7.so.1.0
>\#5  0x4a1c46c1 in PyEval_EvalCodeEx () from /usr/lib/libpython2.7.so.1.0
>\#6  0x4a1c4843 in PyEval_EvalCode () from /usr/lib/libpython2.7.so.1.0
>\#7  0x4a1de369 in ?? () from /usr/lib/libpython2.7.so.1.0
>\#8  0x4a1e0536 in PyRun_InteractiveOneFlags () from /usr/lib/libpython2.7.so.1.0
>\#9  0x4a1e0713 in PyRun_InteractiveLoopFlags () from /usr/lib/libpython2.7.so.1.0
>\#10 0x4a1e0d38 in PyRun_AnyFileExFlags () from /usr/lib/libpython2.7.so.1.0
>\#11 0x4a1f07cc in Py_Main () from /usr/lib/libpython2.7.so.1.0
>\#12 0x08048567 in main ()

Propanu, 2015-04-21:
Yes, the typemaps can be extended to cover type checking and this is a rather easy fix. I think the interface is actually already set up to do this and only needs minor extensions. We'll look into it soon!

Propanu, 2015-12-24:
Fixed with #321.

closed 2015-12-24 by Propanu