eclipse-upm/upm #179: ADXL345 - Not working

ashishdatta, 2015-05-10:
I am using ADXL345 accelerometer. The sample code compiles but no values are being changed. When I do ``` i2cdetect -y -r 6 ``` I get the address of ```0x4c``` but I notice in the src code for adxl the device address is ```0x53``` any thoughts?

Propanu, 2015-05-11:
Hi Ashish,

The ADXL345 is known to act peculiar on the Intel Edison with the Arduino breakout. Is that what you are using? For the Grove version, the only way to get this sensor working was to drop the board to 3.3V. Who's the vendor for yours? I believe the chip also has an address pin that can be used to select between `0x53` and `0x1D`, but `0x4C` doesn't make sense.

ashishdatta, 2015-05-11:
Hi, sorry for lack of information was heading to the airport when I made the issue. 

We are using the Arduino expansion board and we moved the jumper on the board so we get 3.3V. We are using the Grove ADXL345. I confirmed with at least 2 other accelerometers from Grove that they are reading out as ```0x4C``` , it could be that these are a bad batch of accelerometers, because I know that the one I have in my office has the address of ```0x53``` , will double check that is true when I get back. 

Also I test with 2 board one using weekly 120 and the other using weekly 146 build.

Propanu, 2015-05-11:
Thanks for filling in the blanks. When you get the chance also list the revision (version) number on the sensors. Do you get the `0x4C` address only when the sensor is connected? If you've been using the audio drivers, I suspect that might be the address of the PCM codec...

Propanu, 2015-05-13:
Ashish, if you've been using the accelerometer from the new starter kits (±1.5g), they are the MMA7660 and should have an address of `0x55`.

closed 2015-05-13 by Propanu