eclipse-upm/upm #182: Problem of nrf8001

xueliu, 2015-05-15:
Hi,

I have a nrf8001 board form RedBearLab. And this chip is supported by UPM. I have tried the test programs nrf8001-broadcas and nrf8001-helloworld, but they can not work properly. I can not find the ble device from my smartphone. Does anyone have some idea ?

Propanu, 2015-05-15:
Hello,

Could you please give us more details about your setup? Do you have the Adafruit version of the sensor? What Intel IoT device and board are you using?

Thanks

xueliu, 2015-05-15:
Hi,

Thank you for your reply. I don't have the Adadruit version of the sensor. What I have is BLE Shield from RedBearLab. It is working OK with Arduino UNO. 

I have Intel Edison Kit for Arduino.

xueliu, 2015-05-15:
About the setup. I have set the REQN to Pin 10 and RDYN to PIN 8 according to the example code

Propanu, 2015-05-15:
This driver is rather complex and also depends on the acilib library, thus any changes there might break it. I also believe it was developed using the Olimex version of the sensor which might be different from the RBL shield. Sadly, right now I don't have any nrf8001 available, but there are a few things I can suggest.

Since you mentioned it works on the Arduino UNO, try using the same driver with the Arduino SDK (1.6.4) on the Edison. There are also specific drivers from vendors or from Nordic Semiconductor themselves https://github.com/NordicSemiconductor/ble-sdk-arduino. If you can find a driver that works for the shield in Arduino mode on the Edison, then it may be possible to port and merge it with the existing driver.

xueliu, 2015-05-15:
I think no matter what version of the sensor. Both of them use the same nrf8001 chip, so there should be no different.

The driver from Nordic Semiconductor themselves https://github.com/NordicSemiconductor/ble-sdk-arduino can officially support BLE Shield. Every examples can run on Arduino UNO + BLE Shield. Because UPM driver of nrf8001 is also based on ble-sdk-arduino. It is reasonable to believe that UPM should also support BLE Shield.

Unfortunately ble-sdk-arduino can not be compiled in the Arduino SDK (1.6.4) for the Edison.

Propanu, 2015-05-16:
You're right, the UPM driver is basically just a port of ble-sdk-arduino that's using MRAA to setup the pins. Thus the only other things that come to mind, do you get any errors on initialization and what version of MRAA are you using?

xueliu, 2015-05-18:
I have a kernel complain:
```
spidev spi5.1: setup: unsupported mode bits 8
```
from
```
mraa_spi_lsbmode(spi, 0);
```
But it does not affect the SPI bus. Without it there is no error message.

I clone the latest code of MRAA and UPM from github into the Edison and compile and install directly.

Propanu, 2015-05-19:
So did you comment out that line from _hal_aci_tl.cpp_ then? By default MRAA sets up SPI as msb first with
`mraa_spi_lsbmode(dev, 0)` 
The nrf8001 driver also provides a REVERSE_BITS macro which is actually being used in _m_aci_spi_transfer_. This leads me to believe that with the current code the packets are possibly reversed twice leading back to the msb format which is not what the device expects. It's possible to check for this by adding debug code that dumps packet contents to the console.
This would also allow you to reference against the specsheet to see if the device initializes properly and that's the next place I would look into. Unfortunately I won't be able to help with that until I receive this sensor myself.

Propanu, 2015-09-09:
I've updated the list of known limitations with this and unfortunately I don't think it will work on the Intel Edison without a kernel patch addressing the SPI bus. The driver works as expected on Intel Galileo boards though.

closed 2015-09-09 by Propanu