eclipse-upm/upm #203: Node.JS documentation doesn't display arrows next to categories.

Propanu, 2015-06-04:
This is a Yuidoc related issue introduced after the IE 8+ fix, and likely caused either by the style sheet or helper.js. Visible only on external site and unable to reproduce with local builds, which show the arrow properly.

wahbahdoo, 2015-06-04:
I don't think you pulled in all the files correctly manually. 
For example, I looked at the CSS loaded for http://iotdk.intel.com/docs/master/upm/node
and it's not the same as https://github.com/intel-iot-devkit/doxygen2jsdoc/blob/master/generators/yuidoc/tmpl/upm/assets/css/main.css

Propanu, 2015-06-04:
Yes I've noticed that too, and that's the problem. I've done the merge with a diff tool and just checked again now and everything is in order. The next thing to look at is to compare the versions of the node modules that I have installed locally vs. what's on iotdk.

wahbahdoo, 2015-06-04:
hmm and you're sure that you generated the documentation with all the files in the latest https://github.com/intel-iot-devkit/doxygen2jsdoc/blob/master/generators/yuidoc/tmpl/upm ?

Propanu, 2015-06-04:
Yes, this is the commit right here: 0c46dd28beca45a30861bd795c958f1d8ff9344d
All your latest changes were merged and iotdk seems to be running the lates versions for the node modules. This leaves me pretty much clueless as to what might be causing it.

Propanu, 2015-06-09:
Seems like this was resolved after documentation was refreshed on iotdk.

closed 2015-06-09 by Propanu