eclipse-upm/upm #207: HCSR04 doesnt work

QuteBits, 2015-06-14:
Can only take one reading, segfaults on delete, inaccurate.

Propanu, 2015-06-15:
We are aware that this driver needs a rewrite as it should not use the PWM to trigger the sensor. It is on our list and will be addressed soon!

rafaneri, 2015-06-15:
I had the same problem and I'm working on a fix for the sensor. When I finish the tests will submit a pull request to repository. That's between today and tomorrow.

Propanu, 2015-07-06:
@rafaneri are you still working on this?

rafaneri, 2015-07-07:
Yes, I'm still working on this... refactoring the code to adjust mraa_gpio_edge_t and to work with nodejs

rafaneri, 2015-08-20:
Hello @QuteBits,

I did some changes in sensor code, but I need to do more tests before send the code to repository. Can you help me to do some tests too?

In constructor method:
	
```c
	m_pwmTriggerCtx     = mraa_pwm_init (triggerPin);
    if (m_pwmTriggerCtx == NULL) {
        std::cout << "PWM context is NULL" << std::endl;
        exit (1);
    }
```
	
	to
	
```c
	m_triggerPinCtx     = mraa_gpio_init (triggerPin);
    if (m_triggerPinCtx == NULL) {
        fprintf (stderr, "Are you sure that pin%d you requested is valid on your platform?", triggerPin);
        exit (1);
    }
	
	mraa_gpio_dir(m_triggerPinCtx, MRAA_GPIO_OUT);
	mraa_gpio_use_mmaped(m_triggerPinCtx, true) // Use mmaped to echo pin too
```

	getDistance:
	
```c
	mraa_gpio_write(m_triggerPinCtx, 1);
	usleep(10);
	mraa_gpio_write(m_triggerPinCtx, 0);

	m_doWork = 0;
	m_InterruptCounter = 0;
	while (!m_doWork) {
	  usleep (1); // it's a test
	}
	
	double duration = m_FallingTimeStamp - m_RisingTimeStamp;
	
	return (duration/2) / 29.1;
	
```

QuteBits, 2015-08-20:
>
> I did some changes in sensor code, but I need to do more tests before send
> the code to repository. Can you help me to do some tests too?
>
yes - did you make the changes in a GitHub fork/branch? If yes, please
provide me with a link and I will make any changes if needed. Or did you
send me all the changes in the email?

rafaneri, 2015-08-28:
Hello @QuteBits.

[It's](https://github.com/rafaneri/upm) the repository [rafaneri/upm](https://github.com/rafaneri/upm).
I'm having some troubles when running the example code. In the first time, if the sensor already is connected, the "edge" event is not "detected", but if I remove (with the example running) the sensor wires and reconnect the sensor come back to work fine.

I not know how to resolve this problem.

rafaneri, 2015-08-29:
So,

today I did some test and some changes. In sensor constructor I set the trigger pin to 0. And before use the sensor instance I use a delay (1 second) and now it's working.

```c
int
main(int argc, char **argv)
{
    sonar = new upm::HCSR04(5, 6, &interrupt);
    signal(SIGINT, sig_handler);

    sleep(1); // using the delay here

    for(;;){
		std::cout << "get distance" << std::endl;
		double distance = sonar->getDistance(CM);
		std::cout << "distance " << distance << std::endl;
		sleep(5);
	}

    delete sonar;

    return 0;
}
```

Propanu, 2015-08-31:
Great! Can you please open a PR for your changes?

rafaneri, 2015-09-01:
Yes, of course! :)
I did it now!

Propanu, 2015-09-15:
This was patched in #248!

closed 2015-09-15 by Propanu