eclipse-upm/upm #208: Why is this not available in NPM?

yocto1, 2015-06-14:
Hi, is this repo not available in NPM or how can I integrate with NPM.

Thanks in advance

alext-mkrs, 2015-06-15:
Mraa is available in NPM, see https://www.npmjs.com/package/mraa

yocto1, 2015-06-15:
Yeah, I have installed that. But the modules included under upm are not installed, is there anyway to add these with npm?

Propanu, 2015-06-15:
UPM is not available yet in NPM but will be published soon.

alext-mkrs, 2015-06-15:
Oops, sorry - reading this on my phone I overlooked the fact that's a comment for the UPM repo and not the MRAA one :smiley:

yocto1, 2015-06-15:
Yeah no problem, easily done, Having some issues trying to get these installed to use in Node.js. Anyone got any pointers that would help me out now? a simple guide to maybe do it, little stuck as the instructions are a little foggy!

alext-mkrs, 2015-06-15:
Looks like I've already answered in IRC :smiley: but let me repeat here for the history - as long as you're running Edison, the easiest way would actually be to install the MRAA and UPM packages from the opkg repository.

Just run `opkg update && opkg install upm` and it will install UPM and MRAA as a prerequisite.

Propanu, 2015-06-16:
Yes, the way Alex suggested is usually the best way to do it unless you are modifying sensor drivers or trying to run this on a custom platform that doesn't have `opkg`. If you still want to build the library and have issues let us know and we can work on it.

Propanu, 2015-09-25:
UPM 0.4.0 is on NPM now for those using Ubilinux or other custom distros. If you are using the official Yocto image we still recommend using `opkg` instead.

closed 2015-09-25 by Propanu