eclipse-upm/upm #211: adafruitss.cxx frequency is hard coded

wolfmanjm, 2015-06-27:
It appears that the PWM frequency is hard coded here...
https://github.com/intel-iot-devkit/upm/blob/master/src/adafruitss/adafruitss.cxx#L60

Instead of using the passed in frequency. This means that although the frequency may be changed the pulse width will be incorrect for the given frequency.

I'll fix this and test it before submitting a PR for the fix.

Propanu, 2015-06-29:
You might want to take a look at the PCA9685 driver provided by @Jon-ICS. He used it to implement the Adafruit MotorShield, since that one also uses the same PWM driver and might come in handy here.

Jon-ICS, 2015-06-29:
This driver should probably be re-written to use pca9685 :)

wolfmanjm, 2015-06-29:
this particular bug is not related to the PCA9685, it is specific to determining the pulse width for a servo used later on.

I agree using the PCA9685 would make sense though to avoid duplication, but that is simply a refactor :)

wolfmanjm, 2015-07-31:
I've fixed and tested the pwm frequency bug. I'm still looking at how hard it is to use the PCA9685 driver. It initially looks like there are some missing features, but I need to investigate further.

Propanu, 2015-07-31:
Can you please submit a pull request with your fix?

closed 2015-08-06 by Propanu