eclipse-upm/upm #214: UPM no longer builds with latest MRAA due to GPIO enums refactoring

alext-mkrs, 2015-07-05:
Since https://github.com/intel-iot-devkit/mraa/commit/f080a744973d6687129cf2fd5e63175d46a753be GPIO enums have different names, which causes UPM to fail during buildign with errors like

```
<cut>
| /edison_dev/yocto/ww18/build/tmp/work/core2-32-poky-linux/upm/0.3.1+gitAUTOINC+cff584d8b9-r0/git/src/hcsr04/hcsr04.cxx: In constructor 'upm::HCSR04::HCSR04(uint8_t, uint8_t, void (*)(void*))':
| /edison_dev/yocto/ww18/build/tmp/work/core2-32-poky-linux/upm/0.3.1+gitAUTOINC+cff584d8b9-r0/git/src/hcsr04/hcsr04.cxx:52:5: error: 'gpio_edge_t' was not declared in this scope
|      gpio_edge_t edge = MRAA_GPIO_EDGE_BOTH;
|      ^
| /edison_dev/yocto/ww18/build/tmp/work/core2-32-poky-linux/upm/0.3.1+gitAUTOINC+cff584d8b9-r0/git/src/hcsr04/hcsr04.cxx:52:17: error: expected ';' before 'edge'
|      gpio_edge_t edge = MRAA_GPIO_EDGE_BOTH;
|                  ^
| /edison_dev/yocto/ww18/build/tmp/work/core2-32-poky-linux/upm/0.3.1+gitAUTOINC+cff584d8b9-r0/git/src/hcsr04/hcsr04.cxx:53:34: error: 'edge' was not declared in this scope
|      mraa_gpio_isr (m_echoPinCtx, edge, fptr, NULL);
|                                   ^
| make[2]: *** [src/hcsr04/CMakeFiles/jsupm_hcsr04.dir/hcsr04.cxx.o] Error 1
<cut>
```

I think I could prepare the patch for UPM to fix this, but I wonder what's your view on that, i.e. do you assume only released MRAA versions as valid for building UPM? CMake config has `>=` only and I don't see any other mentions of this elsewhere.

Propanu, 2015-07-06:
There are other sensors where interrupts are used, most likely however the enum value is used directly and not by having a variable defined. Simply do the same here and the issue will be resolved.
Renamed data structures can also be handled with version based macros when API changes take place in order to ensure compatibility with a `>=` version when such an easy fix is not available. As of now requiring a minimum version is the only constraint and I would rather keep it & bump whenever a major change happens.

alext-mkrs, 2015-07-06:
Makes sense, thanks - and indeed it's just the case that elsewhere the enums were used directly. In this case, after looking at it more attentively, it really doesn't add any value to have a separate variable, so I've replaced the variable with a direct use, just as you've said. Please see the PR I've sent, builds fine for me.

PR #216 (2015-07-06): hcsr04: Fixed compilation error after MRAA enum renaming
Closes #214.

It also closes #215, which is a duplicate of #214.
commit 08ff5006 (2015-07-06): hcsr04: Fixed compilation error after MRAA enum renaming

Closes #214.
closed 2015-07-06 by Propanu