eclipse-upm/upm #220: Segfault with pca9685 and mraa_i2c_init

tripzero, 2015-07-22:
Python 2.7.9 (default, Jul 20 2015, 13:11:32)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyupm_pca9685
>>> pwm = pyupm_pca9685.PCA9685(0)

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff606abc9 in mraa_i2c_init () from /usr/lib/libmraa.so.0
(gdb) bt
#0  0x00007ffff606abc9 in mraa_i2c_init () from /usr/lib/libmraa.so.0
#1  0x00007ffff62838a5 in upm::PCA9685::PCA9685(int, unsigned char) ()
  from /usr/lib/python2.7/site-packages/_pyupm_pca9685.so
#2  0x00007ffff628084a in ?? ()
  from /usr/lib/python2.7/site-packages/_pyupm_pca9685.so
#3  0x00007ffff7addb1c in PyEval_EvalFrameEx () from
/usr/lib/libpython2.7.so.1.0
#4  0x00007ffff7adebe0 in PyEval_EvalCodeEx () from /usr/lib/libpython2.7.so.1.0
#5  0x00007ffff7a59b3c in ?? () from /usr/lib/libpython2.7.so.1.0
#6  0x00007ffff7a2c143 in PyObject_Call () from /usr/lib/libpython2.7.so.1.0
#7  0x00007ffff7a3fbcd in ?? () from /usr/lib/libpython2.7.so.1.0
#8  0x00007ffff7a2c143 in PyObject_Call () from /usr/lib/libpython2.7.so.1.0
#9  0x00007ffff7a965d7 in ?? () from /usr/lib/libpython2.7.so.1.0
#10 0x00007ffff7a949d7 in ?? () from /usr/lib/libpython2.7.so.1.0
#11 0x00007ffff7a2c143 in PyObject_Call () from /usr/lib/libpython2.7.so.1.0
#12 0x00007ffff7adb92b in PyEval_EvalFrameEx () from
/usr/lib/libpython2.7.so.1.0
#13 0x00007ffff7adebe0 in PyEval_EvalCodeEx () from /usr/lib/libpython2.7.so.1.0
#14 0x00007ffff7aded19 in PyEval_EvalCode () from /usr/lib/libpython2.7.so.1.0
#15 0x00007ffff7afb0ef in ?? () from /usr/lib/libpython2.7.so.1.0
#16 0x00007ffff7afd56a in PyRun_InteractiveOneFlags ()
  from /usr/lib/libpython2.7.so.1.0
#17 0x00007ffff7afd74e in PyRun_InteractiveLoopFlags ()

Propanu, 2015-07-22:
Hi, we never had issues with this driver before, my initial guess is that your PCA9685 address is not set to default 0x60. Can you please provide some more details about your board, setup and sensor vendor?

tripzero, 2015-07-22:
It might be true that my address is wrong.  I'm using the adafruit board.
I'll double check.  However, it shouldn't seg fault even if the address is
incorrect.

rwaldron, 2015-07-22:
> However, it shouldn't seg fault even if the address is incorrect.

I've also encountered this with mraa & upm and completely agree. 

Anyway, adafruit's pca9685 default address is `0x40`

Propanu, 2015-07-24:
So does this work for you if you pass the right address to the constructor then?

tripzero, 2015-07-29:
No.  Passing 0x40 still segfaults.  I might be using the wrong bus for the
MinnowBoard max, but its hard to tell.

Propanu, 2015-08-04:
Maybe permission issue on `/dev/i2c-X`? I recall seeing this somewhere before.

tripzero, 2015-08-04:
I hope it's not a permissions issue.  I'm root!

root@intel-corei7-64:~# ls -l /dev/i2c-*
crw-------    1 root     root       89,   7 Aug  3 15:39 /dev/i2c-7

I discovered most of the seg faults were due to the wrong bus.  The mraa
bus should be "0".  mraa will look for the first bus with "designware" in
the name and assign that as bus "0".  This is always /dev/i2c-7 for me.  If
I use the right bus and the right device id, I can read from the device,
but cannot write:

~~~~
>>> import pyupm_pca9685
>>> pwm = pyupm_pca9685.PCA9685(0, 0x40)
writeByte: mraa_i2c_write_byte() failed.
MRAA: Invalid Handle.
>>> quit()
~~~~

The problem appears to be in mraa_write_byte[_reg]() because I can
reproduce this with straight mraa:

~~~~
root@intel-corei7-64:~# python
Python 2.7.9 (default, Aug  3 2015, 09:36:17)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> i = mraa.I2c(0)
>>> i.address(0x40)
0
>>> i.readReg(0x00)
0L
>>> i.writeReg(0x00, 0x31)
5
~~~~

tripzero, 2015-08-04:
I hope it's not a permissions issue.  I'm root!

root@intel-corei7-64:~# ls -l /dev/i2c-*
crw-------    1 root     root       89,   7 Aug  3 15:39 /dev/i2c-7

I discovered most of the seg faults were due to the wrong bus.  The mraa
bus should be "0".  mraa will look for the first bus with "designware" in
the name and assign that as bus "0".  This is always /dev/i2c-7 for me.  If
I use the right bus and the right device id, I can read from the device,
but cannot write:

```python
import pyupm_pca9685
pwm = pyupm_pca9685.PCA9685(0, 0x40)
```
> writeByte: mraa_i2c_write_byte() failed.
> MRAA: Invalid Handle.
The problem appears to be in mraa_write_byte[_reg]() because I can
reproduce this with straight mraa:

```python
import mraa
i = mraa.I2c(0)
i.address(0x40)
i.readReg(0x00)
(returns: 0L)
i.writeReg(0x00, 0x31)
(returns: 5)
```

Propanu, 2015-08-06:
Not really a lot that can be done from UPM then. The driver is already checking the result of `mraa_i2c_init()` and will complain if it fails as you can see. If you want to open this with MRAA and continue tracking from there please cross-reference to this. Thanks!

closed 2015-08-06 by Propanu