eclipse-upm/upm #225: node.h: No such file or directory compilation terminated.

SteBuc, 2015-08-03:
Hi there, following the build guide on my device when I'm met with the following error when trying to install the individual driver at the section? Any ideas?

root@ubilinux:/home/edison/upm-master/build# make
[  0%] Built target _pyupm_joystick12
[  0%] Built target joystick12
[  0%] Building CXX object src/joystick12/CMakeFiles/jsupm_joystick12.dir/jsupm_joystick12JAVASCRIPT_wrap.cxx.o
/home/edison/upm-master/build/src/joystick12/jsupm_joystick12JAVASCRIPT_wrap.cxx:156:18: fatal error: node.h: No such file or directory
compilation terminated.
make[2]: *** [src/joystick12/CMakeFiles/jsupm_joystick12.dir/jsupm_joystick12JAVASCRIPT_wrap.cxx.o] Error 1
make[1]: *** [src/joystick12/CMakeFiles/jsupm_joystick12.dir/all] Error 2
make: *** [all] Error 2

Propanu, 2015-08-03:
Hi! You might have `node` installed in a non-typical location. What does cmake tell you when you generate the makefiles, was it found?
In such cases, use the `NODE_DIR` environment variable to point to the location where node is installed. Alternatively, if you're not interested in the python/node bindings for upm you can turn them off with -DBUILDSWIG=OFF when you run cmake.

SteBuc, 2015-08-03:
Thanks man,

Get the following from cmake so I assume its found

INFO - Target arch is i586
INFO - Node version is 0.10.40
INFO - Node using v8 3.14.5.9

-- Configuring done

Tried echo $NODE_DIR but not getting anything back however.

Looking to use node a little so would obviously like to keep.

Propanu, 2015-08-04:
`NODE_DIR` is not set by default but you can use it when you know it's installed somewhere else, this shouldn't be the case here though since cmake finds it.
Were you able to build the node bindings for MRAA?
You can try to add `-DCMAKE_INCLUDE_PATH=/usr/include` when you run cmake.

SteBuc, 2015-08-04:
The mraa has been built successfully but I'm not sure if it has had its node bindings configured correctly, how can i verify this?

Tried the above when running cmake and still having the same error

Propanu, 2015-08-04:
You can check your `mraa/build/src/javascript` folder for the `mraa.node` and `package.json` files. I seriously doubt you had `-DBUILDSWIGNODE=ON` for MRAA because then it would work for UPM too.

Can you please try running `make -n` or `make VERBOSE=1` so that we can at least see where it's trying to look for the header?

It's very likely that your node binary is there but the includes are somewhere else or missing, in fact, it's a rather known issue when using Ubilinux (sorry I just noticed now that you're not using Yocto). See this MRAA [issue](https://github.com/intel-iot-devkit/mraa/issues/148). I have no clue if Ubilinux provides `node-dev`, but if they do install it and give it a try.

Otherwise, my suggestion would be to start with node from scratch, and build from sources for your distro. And if your SWIG version is less than 3 build that too.

Propanu, 2016-01-30:
Did you ever resolve this? I plan to close the question as our node builds have changed quite a bit in the last months...

g-vidal, 2016-02-19:
Hi,
I have got the same problem trying to compile on debian stretch I found the workaround using your trick.
node.js has been compiled from sources (to get version 5) and is installed in default location :   

```
zagami.€ : which node
/usr/local/bin/node
zagami.€ : ls /usr/local/include
mraa/  mraa.h  mraa.hpp  node/
```
You may notice that mraa that compiled like a charm is also installed in default /usr/local, following your advice I tried 
`cmake .. -DCMAKE_INCLUDE_PATH=/usr/local/include/ -DBUILDSWIG=OFF`
which works perfectly :-) but unfortunately I would like to get at least python libs... :-( and like a miracle using 
`cmake .. -DCMAKE_INCLUDE_PATH=/usr/local/include/`
works now perfectly meanwhile  my first attempt failed... with the same option.

For me the answer solved the question!

Many thanks

Propanu, 2016-04-11:
Closing as newer versions of the library will detect and use Node.js properly thus this should no longer be an issue. Just remember that you also need the Node.js headers, as the binary alone is not enough to build the bindings.

closed 2016-04-11 by Propanu