eclipse-upm/upm #228: Node upgrade on Edison to v0.12.7 causes: Error: Module did not self-register.

volteoMatt, 2015-08-07:
We recently upgraded our board to v0.12.7 to address a Segmentation Fault we keep getting (understanding is that v12+ fixes this bug). However, after upgrading, we get an error that a module did not self-register. I am thinking this is due to all of the jsupm packages that come pre-installed in /usr/lib/node_modules. I was hoping an npm rebuild would fix, but this fails on the mdns2 package. npm update doesn't work. 

Does anyone know the fix for this?

Here is the full output:

module.js:355
  Module._extensions[extension](this, filename);
                               ^
Error: Module did not self-register.
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/home/root/.node_app_slot/asu_sez.js:3:19)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)

Thank you for any help!

Propanu, 2015-08-07:
Hi,

You will likely have to remove all the node modules and install them again. The API changes from 0.10 to 0.12 are pretty significant. Please keep in mind that UPM is not available with npm yet, so you will have to build the sensor libraries manually.

Good luck!

volteoMatt, 2015-08-07:
Yikes - sounds a bit daunting seeing as there appears to be ~100+ sensor libraries...

Propanu, 2015-08-07:
Heh, it's not really that bad, the build process is fully automated with cmake. See our [build documentation](../blob/master/docs/building.md) for additional information on how it works. All you'd have to do is configure some paths to MRAA and cmake will take care of the rest. Then `make install` and your libraries are generated. If you do this directly on the Edison it's pretty straightforward.

volteoMatt, 2015-08-11:
We were not able to find the paths to MRAA defined in the UPM build documentation. We went through a rebuild of MRAA to see if that put them in the right path, but it did not. Is there anything special we have to do to compile MRAA in the directories/paths specified in the build docs?

Thanks for the help!

Propanu, 2015-08-12:
Those path variables are not set automatically when MRAA builds, it is up to you to define them before you start the UPM build. So for instance, if you don't change the install prefix for MRAA, it will stay `/usr/local`. Now, if you build under the home folder as root, then you would set them like this:

    PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/home/root/mraa/build/usr/local/lib/pkgconfig
    CPLUS_INCLUDE_PATH=$CPLUS_INCLUDE_PATH:/home/root/mraa/build/usr/local/include
    LIBRARY_PATH=$LIBRARY_PATH:/home/root/mraa/build/usr/local/lib

This of course assumes that you cloned and built MRAA there first and followed the build instructions from that repository.

volteoMatt, 2015-08-13:
Got it working! Thanks for all the help. Our segmentation faults have gone away it appears, so the upgrade fixed this issue.

closed 2015-08-13 by volteoMatt