eclipse-upm/upm #234: Unable to use MPR121 driver with Intel Edison

stewart, 2015-08-17:
I'm attempting to communicate with the MPR121 via the UPM node module.

I'm using the Arduino breakout board, and am getting power from the wall.

I've been using some other I2C modules fine (grove base kit display, etc) but can't get the MPR121 sensor to work - here's my test code:

```javascript
var mpr121 = require("jsupm_mpr121");

var touch = new mpr121.MPR121(mpr121.MPR121_I2C_BUS, mpr121.MPR121_DEFAULT_I2C_ADDR);

touch.configAN3944();
// always logs "configAN3944: 101: I2C write failed." and returns false
```

I've tried all I2C ports on the Grove Shield, to no avail. Similarly, the [provided example](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/mpr121.js) fails with the same error.

stewart, 2015-08-17:
Similarly, trying to connect a [TSL2561](http://www.seeedstudio.com/wiki/Grove_-_Digital_Light_Sensor) digital light sensor, per the [provided example](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/tsl2561.js):

```
> var light = new (require("jsupm_tsl2561").TSL2561)();
Error: on i2c bus write reg in i2cWriteReg()
Error: Unable to power up - Ensure TSL2561 is connected to I2C
```

Propanu, 2015-09-09:
Unfortunately, the MPR121 revision 1.3 is known to be incompatible with the Intel Edison Arduino board. Older versions of this sensor from Seeed work fine though. The TSL2561 apparently drops reads/writes now and then with the Intel Edison Arduino board. Usually it's enough to cause it to fail on initialization using the UPM driver, but I was able to get some data out of it manually with `i2cset` & `i2cget`. It will work with the mini-breakout and I've updated the list of known limitations to reflect this.

Thanks!

closed 2015-09-09 by Propanu