eclipse-upm/upm #236: Stepper Motors are Synchronous

stewart, 2015-08-18:
This may be by design, but in Node.js the stepper motor controls are synchronous, and completely block execution.

e.g.

```javascript
var ULN200XA = require("jsupm_uln200xa");

var step1 = new ULN200XA.ULN200XA(4096, 8, 9, 10, 11),
    step2 = new ULN200XA.ULN200XA(4096, 4, 5, 6, 7);

function move(stepper, dir, speed, steps) {
  stepper.setSpeed(5);
  stepper.setDirection(dir || CLOCKWISE);
  stepper.stepperSteps(steps || 1020);
}

// stepper 2 will not start until stepper 1 is completely finished turning
move(step1);
move(step2);
```

This also severely complicates IoT use cases, as while the motors are turning, the rest of the program is unresponsive (e.g. servers, network requests, etc)

rwaldron, 2015-08-18:
Everything in upm is process blocking/synchronous. The mraa io operations themselves are synchronous, but this library does nothing to mitigate that, eg. breaking up reads and writes over many turns of the event loop.

stewart, 2015-08-18:
@rwaldron yeah, I'd noticed this as well - but it's proving particularly troublesome with the stepper motors, since a lot of uses require them to be operated in parallel.

Propanu, 2015-09-09:
Hi, sorry for the delayed response.

This is a great and interesting topic for us, but I'm not sure if UPM is best place to have it, it's more of a MRAA thing.

As far as I remember, when we looked into this for MRAA, sync vs. async really makes no difference at least performance wise. On the other hand, C/C++, Python and JAVA APIs are generally synchronous and concurrency has to be implemented by the user. Sure, there are exceptions to the rule and all these languages offer some form of async invocation, but they are explicit and language specific. I hope you get the idea why this would be a bad idea to implement at UPM level.

Further, as @rwaldron mentioned, we use SWIG for our API generation from native C/C++ code and while SWIG is a great tool, it's almost impossible to deal with advanced concepts without manual intervention and custom interface files. Threads fall into this category and while you'll find that some UPM drivers use them internally, it's not straightforward to expose a running context across bindings to another language. Directors provide a workaround for Python and Java, enabling us to at least use callbacks, but they are unavailable for Node.js. Just look at the hack @arfoll had to implement to enable them in MRAA for Node.js using libuv threads.

Anyway, before this becomes a TL:DR answer, let me get to the point of solving your problem. I initially looked at Caolan's popular *async* module, but after a bit of reading realized that when using `async.parallel()` *"JavaScript remains single-threaded"*. So instead I turned to *webworker-threads* and *threads-a-gogo*. Surprise, I can't load modules in spawned threads and I also have to do all the serialization manually. That's how I ended up with *parallel.js* which works quite nicely and doesn't suffer from the two issues (as long as the data you pass in is valid JSON).

```JavaScript
var Parallel = require('paralleljs');

var move = function (data) {
    var ULN200XA = require("jsupm_uln200xa");
    var stepper = new ULN200XA.ULN200XA(4096, data[0], data[1], data[2], data[3]);
    stepper.setSpeed(6);
    stepper.setDirection(ULN200XA.ULN200XA.DIR_CW);
    stepper.stepperSteps(data[4] || 1024);
    stepper.release();
    return null;
};
 
var step1 = new Parallel([8, 9, 10, 11, 2048]);
var step2 = new Parallel([4, 5, 6, 7]);
 
// Spawn our sync functions in parallel
step1.spawn(move).then(function(){console.log("Stepper 1 done")});
step2.spawn(move).then(function(){console.log("Stepper 2 done")});
```

Now I know this does require some extra code and passing in seemingly random values doesn't make a lot of sense at first, but you can wrap thread creation with more descriptive functions if needed.

I'll leave this open as it might help others trying to implement concurrency in Node.js using our libraries! And really, with the Intel IoT Dev Kit you are only limited by your imagination :wink:

stewart, 2015-09-09:
I'm no longer working on the project to which this issue related, but I appreciate the feedback and am sure someone will find this useful. @deadprogram might also be interested in this information.

closed 2016-04-11 by Propanu