eclipse-upm/upm #243: DS18B20 support

akulo, 2015-08-26:
I would like to see support for One wire DS18B20 Temperature Sensor

Propanu, 2015-09-05:
This is a popular sensor that unfortunately won't end up in UPM any time soon. One Wire is difficult to implement on a non-RTOS! It might become possible with a newer kernel though.

Propanu, 2015-12-08:
Update: We found a way that might allow us to control one wire devices on our Linux boards and a driver for this sensor is under development.

akulo, 2015-12-08:
thats a great news! looking forward to the updated driver.

alext-mkrs, 2016-01-25:
I guess this pull request for mraa should open the door for UPM to support it, but not earlier than it's merged: https://github.com/intel-iot-devkit/mraa/pull/415

Propanu, 2016-02-02:
Unfortunately this won't make it into 0.5.0 but I will merge the implementation (#340) once MRAA adds the necessary functions.

Propanu, 2016-05-03:
Added to with e679d40d44bd5fb4c71f8567bfe9ceb7f45c75c7!

closed 2016-05-03 by Propanu