eclipse-upm/upm #257: Compile errors with g++ v5.1

whbruce, 2015-09-16:
The new mraa USB sub-platform feature means that you can now develop UPM modules and examples on a Linux workstation as you can access sensors over the FTDI FT4222 USB to i2c bridge.

I am doing this on a Fedora 22 workstation but am hitting compile errors due to the stricter assignment checking in g++ v5. There are many instances of a mraa::Result enum being assigned to mraa_result_t. An example is in src/th02.cxx line 48

    mraa_result_t ret = m_i2c.address(m_addr);
    if (ret != MRAA_SUCCESS) {
        throw TH02Exception ("Couldn't initialize I2C.");
    }

This generates the error

/path/to/upm/src/th02/th02.cxx:48:45: error: cannot convert ‘mraa::Result’ to ‘mraa_result_t’ in initialization
     mraa_result_t ret = m_i2c.address(m_addr);

As I can't find a compiler flag that allows this assignment there are two solutions:
1. Quick hack
    mraa_result_t ret = reinterpret_cast<mraa_result_t>(m_i2c.address(m_addr));
    if (ret != MRAA_SUCCESS) {
        throw TH02Exception ("Couldn't initialize I2C.");
    }

2. Fix code
    mraa::Result ret = m_i2c.address(m_addr);
    if (ret != mraa::SUCCESS) {
        throw TH02Exception ("Couldn't initialize I2C.");
    }

There are about 95 instances of this less than stellar coding pattern just in the src folder alone and I don't want to roto-till the code base. 

Advice welcome.

Henry

Propanu, 2015-09-16:
Hi Henry,

Just pull in the latest changes, we've already applied the fix for the API changes and the new C++ types introduced in MRAA (TH02 was done in b8835958e2e5be450aa9a7a4f04933786b29f03f). A lot of sensors are being transitioned from the old style C API in the process. It might take a few days before things settle down and the build is stable again but I'm compiling without issues on latest MRAA head.

By the way, do you use any special flags to compile with g++ 5? If so it would be nice to integrate them into cmake.

whbruce, 2015-09-16:
I synced my fork and now get a clean build. 
Sorry - I should have done this before creating issue. 
I don't use any special compiler flags.
Closing issue. 
You rock!

Henry

closed 2015-09-16 by whbruce