eclipse-upm/upm #259: UPM doesn't build anymore...

arfoll, 2015-09-18:
There is an issue with SWIG bindings + st7335, you're using mraa_gpio_init but then storing the result in a mraa::Gpio not a mraa_gpio_context. Change comes from ab730038fdb7e8e60be5f45a04b5b2d44131f842 from @andreivasiliu2211.

You really should look into setting up travis so you catch these before merging.

andreivasiliu2211, 2015-09-18:
Hi.

There were some issues regarding the previous merge. I have fixed the
compilation issues. They are now, together with the other proposed changes,
in a pull request here [1], waiting to be merged.

Regards,
Andrei

[1] https://github.com/intel-iot-devkit/upm/pull/256

Propanu, 2015-09-18:
PR was merged and the build is back to normal thanks to @andreivasiliu2211 for patching the remaining sensors. Seems like I was linking against stale objects locally when I tested before pushing.
I'm pretty sure travis wouldn't have made any difference here but yes, it would be nice to have it.

closed 2015-09-18 by Propanu