eclipse-upm/upm #269: RFR359F objectDetected() function always returning true

deadprogram, 2015-09-24:
When running the provided example code located at https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/rfr359f.js, the `objectDetected()` functions is always returning `true`, hence the program always reports `Object Detected`.

I have tried with both the current release version of MRAA/UPM, as well also the `intelgalactic-dev` opkg repo.  I have also tried a different RFR359F on a different Edison board, as well as trying a different pin. Same results.

```
root@edison:~# opkg install libmraa0                                                            
Installing libmraa0 (0.7.3.30) on root.
Downloading http://iotdk.intel.com/repos/1.5/intelgalactic-dev/libmraa0_0.7.3.30_i586.ipk.
Configuring libmraa0.
root@edison:~# opkg install upm     
Installing upm (0.3.2.25) on root.
Downloading http://iotdk.intel.com/repos/1.5/intelgalactic-dev/upm_0.3.2.25_i586.ipk.
Configuring upm.
```

Propanu, 2015-09-28:
Hi Ron,
From my experience the IR Distance Interrupter (RFR359F) works really well and we actually used it for several projects and demos that are available on `intel-iot-devkit`. There's a linear pot on the sensor that lets you adjust sensitivity but the most importantly the red LED on the sensor will let you know if there's something in the way or not. I believe the max range on it is about 1".
Also, Grove has an IR Reflective (RPR220) sensor that looks very similar but performs a different function and this one will stay at a logic 1 if there's nothing in the way.

deadprogram, 2015-09-28:
I am indeed certain I have the correct sensor:

[image]

Propanu, 2015-09-28:
Just checked and the Node.js sample works correctly on MRAA 0.8.0 & UPM 0.4.0, most likely older versions too. So I don't really know what could be causing your issue, maybe you got a bad sensor.

deadprogram, 2015-09-28:
I tried 2 different Edison with 2 different RFR359F's, so unless all of the ones I have are bad?

Propanu, 2015-09-28:
Something else might be going on there. Can you confirm that you see the LED on the sensor turn on/off when you bring it close to an object? I would also recommend bypassing the driver and using the MRAA GPIO example that reads a pin.

deadprogram, 2015-09-28:
Good idea on MRAA, hooking up an Edison right now to try that.

deadprogram, 2015-09-28:
A couple of my RFR359F appear to be bad, but at least 1 of them is working as expected after adjusting. The other RFR359F's I have are always detecting no matter what adjustments. Kind of a "finicky" device, but nothing to do with UPM. Closing, thanks for the support.

closed 2015-09-28 by deadprogram