eclipse-upm/upm #272: pyupm_rotaryencoder returns decreasing values

gaiar, 2015-09-27:
I have RotaryEncoder v1.1 from [Grove Indoor Environment Kit](http://www.seeedstudio.com/depot/Grove-Indoor-Environment-Kit-for-Intel-Edison-p-2427.html).

I tried to use it with code example from [IoT DevZone](https://software.intel.com/en-us/iot/hardware/sensors/grove-rotary-encoder), but for some reasons encoder returns me only decreasing values in both rotating directions. My encoder is connected to D2 pin.

[image]

I asked Intel community for [help](https://communities.intel.com/message/335997#335997), for seems it is not common situation. So I think it is either something in upm library, or I have broken piece of hardware.

Propanu, 2015-09-28:
Hi, I only have v1.0 for the Rotary Encoder and the examples work fine with that. I do recall that the interrupt edge mode was set to rising only for compatibility with both Galileo and Edison, and that's not ideal. Also, there is no HW or SW debouncing done, that would have to be addressed in the application. A side effect of this was that sometimes, at the end of a counterclockwise turn, the position would change in the other direction by 1 or 2 units. Nothing near what you're showing though, as while turning you would still get the right values.

It might be that v1.1 was changed and behaves differently. I will try to get my hands on one and look into this further!

gaiar, 2015-09-28:
Can I help you somehow? Get any from controller?

Propanu, 2015-09-28:
If you happen to have an oscilloscope you could take a look at the output on D2 & D3 to see if they match this:

[image]

gaiar, 2015-09-29:
Unfortunately, I wasn't able to find any available oscilloscope right now. Let me know, if I can be useful somehow else.

Propanu, 2015-09-29:
No worries. Still waiting on my Indoor Environmental Kit, I will let you know what's going on once I get it.

Propanu, 2015-10-01:
Sorry but one I have in the Environmental Kit is also v1.0. Thus I will have to order a new one and that will probably take a few days.

Propanu, 2015-10-16:
Hi, I got the v1.1 encoder, it works just fine and I figured out what your issue is. You are not getting a signal on the second pin of the rotary encoder. Hence if you initialize the sensor on D3 & D4 for instance, but you connect the encoder on pins D2 & D3 on the board, you'll see the exact behavior you described. If the pins are matching in your case there must be something wrong with either the encoder or the wire. Hope this helps!

closed 2015-10-16 by Propanu