eclipse-upm/upm #280: JHD1313M1 not initializing correctly when Edison board powered over USB

petreeftime, 2015-10-02:
After booting the Edison board on USB power, on first run of a program requiring the JHD1313M1 LCD, the LCD does not display the first letter on the first line and the second line. On second run the LCD initializes correctly and both lines are displayed completely.

If Edison is powered using a power adapter the issue does not happen at all.

Propanu, 2015-10-21:
Yes, the RGB LCD is a 5V device and it does rely on an external power supply to work properly. I will add this issue to the documentation. When the board is powered through USB it really depends on the port, some computers/chargers will actually provide enough current for the LCD to function properly while others won't.

alext-mkrs, 2015-12-24:
I've just came across this as well. USB-powered Edison with Grove LCD, using Python. After the board power-on, running the program once results in a second row not being initialized, only the first one works in full. Second and subsequent runs of the same program initialize it properly, until the whole board is powered off (just unplugging the LCD is not enough).

As a workaround I've added double initialization to my code, which looks weird, but works. Because of this I tend to think it may be fixed just by changing timings in https://github.com/intel-iot-devkit/upm/blob/master/src/lcd/jhd1313m1.cxx - my guess is that on USB power it gets just slightly less than necessary current or voltage and because of that the controller init runs slower or something along these lines.

I'll see if I have time to play with that and propose a patch, but you may be faster.

Workaround (Python) that worked for me:
```python
grove_lcd = lcd.Jhd1313m1(0)
del self.grove_lcd
time.sleep(0.5)
grove_lcd = lcd.Jhd1313m1(0)
```

alext-mkrs, 2016-01-18:
Looks like it's not that easy as just those timings :smiley: I've checked that even increasing all timings in the constructor four-fold doesn't help it (and they're actually well above the spec limits anyway).

The voltage on the power pins of the LCD is 4.59V in my setup - barely above the spec'd 4.50V as a minimum.

I'll be looking further into this.

Propanu, 2016-01-30:
From my experience it will work OK but with washed out letters around the 4.74V that you get on a decent USB port. I don't recall if I had to initialize it twice but it's 'usable' and I got no missing letters ever since we patched the write function with a short delay. Once I plug in the external adapter into the board my reference jumps to about 4.97V and the screen is crystal clear without any issues whatsoever.

Nevertheless, the issue was added to our list of [known limitations](../blob/master/docs/knownlimitations.md)

closed 2016-01-30 by Propanu