eclipse-upm/upm #290: Use Grove Ultrasonic ranger with Nodejs and Python

badr-badri, 2015-10-11:
HI guys,
Is there any way to use the ultrasonic ranger sensor with python or node.js ?
Thanks in advance

Propanu, 2015-10-13:
Hi,
We have a driver for the HSCR-04, URM37 and MaxSonar that work with all the supported languages. There was also a contribution with a driver for the Grove version of the ultrasonic sensor but it hasn't been merged yet because it needs some clean-up. Which one do you have?

badr-badri, 2015-10-13:
Thanks for your reply.
I have this one :
http://www.gotronic.fr/art-telemetre-a-ultrasons-grove-sen10737p-18976.htm
I hope that you have a solution for that one.
Thanks again.

Propanu, 2015-10-16:
That's the one. Here's the code: https://github.com/arcatdmz/upm/commit/0277b7fcd7806a5ff11cb9a7eb36c650f3384859 in case you want to build the sensor driver yourself until we merge it.

Propanu, 2015-12-03:
The ultrasonic ranger sensor was added to the main branch. I do expect it to be merged with HSCR04 at some point though since they work the same way.

closed 2015-12-03 by Propanu