eclipse-upm/upm #295: dont work for intel galileo gen 1

kobymaster, 2015-10-15:
var myLEDController_obj = new LEDController_lib.PCA9685(I2CBus, I2CAddr);
^
Error: Illegal arguments for construction of _exports_PCA9685
at Object. (/home/root/prueba1.js:27:27)
at Module._compile (module.js:456:26)
at Object.Module._extensions..js (module.js:474:10)
at Module.load (module.js:356:32)
at Function.Module._load (module.js:312:12)
at Function.Module.runMain (module.js:497:10)
at startup (node.js:119:16)
at node.js:935:3

malikabhi05, 2015-10-20:
Hi kobymaster,

You probably are using the default i2c address in the constructor call. You would have to figure out the address for your device. You can use tools such as i2cdetect to figure out the address. The API documentation for this is at: http://iotdk.intel.com/docs/master/upm/node/classes/pca9685.html

xbolshe, 2015-10-31:
A part of upm/examples/javascript/pca9685.js:

[image]

This problem is not related with an incorrect I2C address. It seems it is related with an incorrect function wrapping in pca9685 node-js module.

Propanu, 2015-11-10:
@xbolshe is this with SWIG 3.0.7 and Node 4?

xbolshe, 2015-11-10:
With SWIG 3.0.8 and Node 4.1

xbolshe, 2015-11-10:
Same with Node 5.0.0

[image]

xbolshe, 2015-11-10:
Hm... syslog makes the problem more clear: I2C address seems bad.

[image]

malikabhi05, 2015-11-10:
Hi, if the i2c address is correct then it should work as expected in all languages, it has been tested and it works with node 0.10.36 and swig 3.0.5

closed 2015-11-10 by malikabhi05