eclipse-upm/upm #300: Pulse sensor in eclipse c++ supported sensor list

Mujaddadi, 2015-10-29:
Hi,
     Not sure it is the right place to post it but posting anyway. Pulse sensor from seeedstudio is supported in upm but not showing in the eclipse c++ supported sensor option. I checked the upm xml file and found it has incomplete definition

 <sensor briefdesc="Pulsensor Pulse Sensor library" flag="upm-pulsensor" link="http://iotdk.intel.com/docs/upm/v0.4.0/group__pulsensor.html"/>

I added this into the xml file but got "sensor data is out of sync please click here to update it" and after updating it get backs to its original file.   

       <sensor briefdesc="Pulsensor Pulse Sensor library" category="medical" connection="AIO" detaileddesc="This is a library for a 3-wire pulse sensor sold by several manufacturers. Usually, you can identify the sensor based on the round breakout and the distinctive heart symbol. callback data" flag="upm-pulsensor" id="pulsensor" link="http://iotdk.intel.com/docs/upm/v0.4.0/group__pulsensor.html" manufacturer="Seeedstudio" name="Pulsensor" wiki="http://www.adafruit.com/products/1093"/>
	   <header connection="GPIO" link="http://iotdk.intel.com/docs/upm/v0.4.0/pulsensor_8h_source.html" name="pulsensor.h"/>
	</sensor>

any solution for it so that this sensor can show in the supported sensor list in eclipse?

Propanu, 2015-11-09:
The parser used by the Eclipse sensor plugin needs some adjustments. I've notified the responsible party about this.

Propanu, 2016-01-12:
Fixed with 5e37830bb7851eeb1579ed6466aa447d7ffb0593. The changes will become visible on the next UPM version tag and Eclipse will be able to pull in the sensor info. You should be able to use the sensor as expected once that happens. Thanks for reporting this!

Mujaddadi, 2016-02-13:
Thanks for the fixed. I updated the libraries in eclipse to the latest version but I am still not able to see the sensor. Any idea why is that?

Mujaddadi, 2016-02-13:
or if is there any way to add it manually, please guide me.

Propanu, 2016-02-17:
What version of UPM did you update to with Eclipse?

Mujaddadi, 2016-02-17:
I am not sure which version it is but I used the update Libraries version and it updated to the latest one. Is there anyway to check the UPM version in eclipse?

Mujaddadi, 2016-02-17:
I found the preference option and it is using MARAA 0.9.0 and UPM 0.4.1

Mujaddadi, 2016-02-17:
I run the c++ example given in upm example and now I am getting this error "symbol lookup error: /tmp/PulseSensor: undefined symbol: _ZN3upm9PulsensorC1EPFv9clbk_dataE  logout"

Propanu, 2016-02-17:
That's expected, the sample was updated to match the new API that was introduced in UPM 0.5.0. If you select custom repository from the options tab and use this link: http://iotdk.intel.com/repos/3.0/intelgalactic/opkg/i586/ it will grab the latest versions and the sample should work.

Mujaddadi, 2016-02-18:
I used that link and updated the libraries. I also had to update the libraries In Galileo to match with eclipse. The sensor was showing in the sensor support list in eclipse but today when I opened the eclipse today, it disappeared again from the list. I rolled back and install the libraries again but still no success. Is that cache issue now?

Mujaddadi, 2016-02-18:
Could you tell me how to add the sensor in my project without the clicking the sensor support option?

Mujaddadi, 2016-02-18:
[image]

Mujaddadi, 2016-02-18:
I think I got what is the issue. Last night I deleted the Eclipse and install the fresh copy. I updated the Marra and upm libraries from the repository link you gave me. Sensor was not listed on the sensor support view. Then I selected the Intel development repository and which rolled back to that repositories. Sensor started listed again in the supported sensor list. Finally I updated again back to the repository through the link which you gave me and it is still the sensor support view. So the problem is somewhere between the updated and eclipse old libraries.

Mujaddadi, 2016-02-18:
Finally I run the sample code and it is working now. Getting these values
callback data (1)
callback data (0)
callback data (1)
callback data (0)

Propanu, 2016-02-18:
Sounds like Eclipse handles this by pulling the xml files + drivers when you use the default repository, but only drivers when you use a custom link. Not much you can do about that other than wait for a new Eclipse version to be released which will have the updated links by default. You can add a sensor manually by adding the corresponding `#include` in your code and the linker flag under `Project -> Properties -> C/C++ Build -> Settings -> Cross G++ Linker -> Libraries`.

Not sure what you mean by proper pulse values, the driver has several variables used internally to determine when a heart beat happens (like bpm, threshold, so on) but they are not exposed. If you think they should feel free to start a new UPM Issue requesting a driver update with a list of features you would like to see.

Propanu, 2016-08-23:
This should be no longer an issue with new versions of Intel® System Studio.

closed 2016-08-23 by Propanu