eclipse-upm/upm #302: Lcm1602 constructor should have a default address

stefan-andritoiu, 2015-11-02:
Current constructor: 
Lcm1602(int bus, int address, bool isExpander=true, uint8_t numColumns = 16, uint8_t numRows = 4);
should have a default address. I propose 0x27, taken from the example file, but I do not have the sensor to test this.

Propanu, 2015-11-09:
This is a very popular part that is available from several manufacturers and the I2C address will usually be different. Using a default value might make some users forget to look for the right address before using the driver and confuse them about why it's not working.

Propanu, 2016-06-01:
I've added the default 0x27 value as you suggested since it's very common with several manufacturers. See de986764d4c84217e979f84c7a8956cadc69fd24, thanks!

closed 2016-06-01 by Propanu