eclipse-upm/upm #312: Why my UPM sensor not on the list ?

neuberfran, 2015-12-02:
I bought this driver http://www.mindsensors.com/search?controller=search&orderby=position&orderway=desc&search_query=smartdrive&submit_search

and create a upm sensor to that: https://drive.google.com/file/d/0B9gPdWq4OH-XM2lFdVhMbU5Hekk/view

in future I will TRY do this: https://www.youtube.com/watch?v=tQX34zQGPtw using windshield wiper motor (instead of LED BlinkM)

Please help me

Propanu, 2015-12-03:
Hi, I don't see your pull request, could you please make sure it's open against the main intel-iot-devkit/upm repository? Thanks!

neuberfran, 2015-12-03:
Whats steps do you follow to pull request new  UPM/Sensor ??

sasmita, 2015-12-03:
Please make sure, your changes are committed in a new branch. You can click on "new pull request" button in the repo header in https://github.com/intel-iot-devkit/upm.

Propanu, 2016-01-11:
Do you need additional help with this?

Propanu, 2016-04-07:
Added with #371!

closed 2016-04-07 by Propanu