eclipse-upm/upm #317: GroveMD.configStepper doesn't work

sticeap, 2015-12-12:
When I try to use jsupm_grovemd I receive an error which say **Object #<GroveMD> has no method 'configStepper'**

I used a sample example:

```javascript
var groveMotorDriver = require('jsupm_grovemd');

var motor = new groveMotorDriver.GroveMD(groveMotorDriver.GROVEMD_I2C_BUS,groveMotorDriver.GROVEMD_DEFAULT_I2C_ADDR);

if(motor){
    motor.configStepper(200, groveMotorDriver.STEP_MODE1);
    motor.setStepperSteps(100);
    motor.enableStepper(groveMotorDriver.GroveMD.STEP_DIR_CW, 10);
}

```

Also all 4 leds for direction from Grove I2C Motor Driver are open. 
What I made wrong there?

Thank you.

Propanu, 2015-12-15:
Hi, most likely you are using an old version of UPM. Stepper support for the Grove Motor Driver and the `configStepper` function was added with v0.4.0.

sticeap, 2015-12-16:
Thank you very much. I updated upm and mraa library and now it's fine.

closed 2015-12-16 by sticeap