eclipse-upm/upm #318: sx1276 and adxrs610 modules: compilation problem

xbolshe, 2015-12-13:
SWIG: 3.0.8
Target board: Intel Galileo Gen1/Gen2
GCC: 4.8.3

DEBUG: Executing shell function do_compile
-- checking for module 'mraa>=0.8.0'
--   found mraa, version 0.8.1
INFO found mraa version: 0.8.1
INFO - UPM Version v0.4.1
INFO - Target arch is i586
INFO - Package arch is i586
INFO - Node version is 5.2.0
INFO - Node using v8 4.6.85.31

-- checking for one of the modules 'libopenzwave'
-- Configuring done
-- Generating done
-- Build files have been written to: /sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/git/build
[  0%] Built target bma220
[  0%] Built target jsupm_bma220
[  1%] Built target grovewater
[  2%] Built target jsupm_grovewater
[  3%] Built target jsupm_stepmotor
[  3%] Built target stepmotor
[  3%] Built target jsupm_wheelencoder
[  4%] Built target wheelencoder
[  4%] Built target groveelectromagnet
[  4%] Built target jsupm_groveelectromagnet
[  4%] Built target ecs1030
[  5%] Built target jsupm_ecs1030
[  5%] Built target jsupm_mq303a
[  5%] Built target mq303a
[  5%] Built target jsupm_xbee
[  5%] Built target xbee
[  5%] Built target ad8232
[  5%] Built target jsupm_ad8232
[  5%] Built target at42qt1070
[  6%] Built target jsupm_at42qt1070
[  6%] Built target jsupm_zfm20
[  6%] Built target zfm20
[  6%] Built target jsupm_m24lr64e
[  7%] Built target m24lr64e
[  8%] Built target jsupm_mlx90614
[  8%] Built target mlx90614
[  8%] Built target grove
[  8%] Built target jsupm_grove
[  8%] Built target jsupm_tsl2561
[  8%] Built target tsl2561
[  9%] Built target jsupm_lol
[  9%] Built target lol
[ 10%] Built target jsupm_max44000
[ 10%] Built target max44000
[ 10%] Built target htu21d
[ 10%] Built target jsupm_htu21d
[ 10%] Built target h3lis331dl
[ 11%] Built target jsupm_h3lis331dl
[ 11%] Built target grovewfs
[ 11%] Built target jsupm_grovewfs
[ 11%] Built target adxl345
[ 11%] Built target jsupm_adxl345
[ 11%] Built target jsupm_pn532
[ 11%] Built target pn532
[ 11%] Built target jsupm_mma7455
[ 11%] Built target mma7455
[ 12%] Built target jsupm_ldt0028
[ 12%] Built target ldt0028
[ 12%] Built target jsupm_l298
[ 12%] Built target l298
[ 13%] Built target jsupm_urm37
[ 13%] Built target urm37
[ 13%] Built target grovevdiv
[ 13%] Built target jsupm_grovevdiv
[ 13%] Built target jsupm_lpd8806
[ 13%] Built target lpd8806
[ 13%] Built target groveeldriver
[ 14%] Built target jsupm_groveeldriver
[ 14%] Built target grovescam
[ 14%] Built target jsupm_grovescam
[ 14%] Built target flex
[ 14%] Built target jsupm_flex
[ 14%] Built target jsupm_pulsensor
[ 14%] Built target pulsensor
[ 14%] Built target grovespeaker
[ 14%] Built target jsupm_grovespeaker
[ 14%] Built target hmtrp
[ 14%] Built target jsupm_hmtrp
[ 15%] Built target jsupm_max31723
[ 15%] Built target max31723
[ 15%] Built target jsupm_servo
[ 15%] Built target servo
[ 15%] Built target jsupm_si114x
[ 16%] Built target si114x
[ 16%] Built target adafruitss
[ 17%] Built target jsupm_adafruitss
[ 17%] Built target jsupm_uln200xa
[ 17%] Built target uln200xa
[ 17%] Built target groveultrasonic
[ 18%] Built target jsupm_groveultrasonic
[ 18%] Built target jsupm_waterlevel
[ 18%] Built target waterlevel
[ 18%] Built target jsupm_micsv89
[ 19%] Built target micsv89
[ 19%] Built target isd1820
[ 19%] Built target jsupm_isd1820
[ 20%] Built target jsupm_ppd42ns
[ 21%] Built target ppd42ns
[ 21%] Built target groveo2
[ 22%] Built target jsupm_groveo2
[ 22%] Built target hm11
[ 22%] Built target jsupm_hm11
[ 22%] Built target jsupm_mic
[ 22%] Built target mic
[ 22%] Built target jsupm_lm35
[ 22%] Built target lm35
[ 22%] Built target cjq4435
[ 23%] Built target jsupm_cjq4435
[ 23%] Built target bmpx8x
[ 24%] Built target jsupm_bmpx8x
[ 25%] Built target jsupm_mpu9150
[ 25%] Built target mpu9150
[ 25%] Building CXX object src/sx1276/CMakeFiles/jsupm_sx1276.dir/jsupm_sx1276JAVASCRIPT_wrap.cxx.o
In file included from /sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/git/build/src/sx1276/jsupm_sx1276JAVASCRIPT_wrap.cxx:1961:0:
/sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/git/src/sx1276/./sx1276.h:90:38: error: 'constexpr' needed for in-class initialization of static data member 'const double upm::SX1276::FXOSC_FREQ' of non-integral type [-fpermissive]
     static const double FXOSC_FREQ = 32000000.0;
                                      ^
/sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/git/src/sx1276/./sx1276.h:94:38: error: 'constexpr' needed for in-class initialization of static data member 'const double upm::SX1276::FXOSC_STEP' of non-integral type [-fpermissive]
     static const double FXOSC_STEP = 61.03515625;
                                      ^
src/sx1276/CMakeFiles/jsupm_sx1276.dir/build.make:67: recipe for target 'src/sx1276/CMakeFiles/jsupm_sx1276.dir/jsupm_sx1276JAVASCRIPT_wrap.cxx.o' failed
make[2]: *** [src/sx1276/CMakeFiles/jsupm_sx1276.dir/jsupm_sx1276JAVASCRIPT_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:5259: recipe for target 'src/sx1276/CMakeFiles/jsupm_sx1276.dir/all' failed
make[1]: *** [src/sx1276/CMakeFiles/jsupm_sx1276.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
WARNING: exit code 2 from a shell command.
ERROR: Function failed: do_compile (log file is located at /sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/temp/log.do_compile.18191)

...........
...........

[ 70%] Building CXX object src/adxl335/CMakeFiles/jsupm_adxl335.dir/adxl335.cxx.o
[ 70%] Linking CXX shared module jsupm_adxl335.node
[ 70%] Built target jsupm_adxl335
Scanning dependencies of target grovecollision
[ 70%] Building CXX object src/grovecollision/CMakeFiles/grovecollision.dir/grovecollision.cxx.o
[ 70%] Linking CXX shared library libupm-grovecollision.so
[ 70%] Built target grovecollision
[ 70%] Swig source
Scanning dependencies of target jsupm_grovecollision
[ 70%] Building CXX object src/grovecollision/CMakeFiles/jsupm_grovecollision.dir/jsupm_grovecollisionJAVASCRIPT_wrap.cxx.o
[ 70%] Building CXX object src/grovecollision/CMakeFiles/jsupm_grovecollision.dir/grovecollision.cxx.o
[ 70%] Linking CXX shared module jsupm_grovecollision.node
[ 70%] Built target jsupm_grovecollision
Scanning dependencies of target adxrs610
[ 70%] Building CXX object src/adxrs610/CMakeFiles/adxrs610.dir/adxrs610.cxx.o
[ 70%] Linking CXX shared library libupm-adxrs610.so
[ 70%] Built target adxrs610
[ 70%] Swig source
Scanning dependencies of target jsupm_adxrs610
[ 70%] Building CXX object src/adxrs610/CMakeFiles/jsupm_adxrs610.dir/jsupm_adxrs610JAVASCRIPT_wrap.cxx.o
In file included from /sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/git/build/src/adxrs610/jsupm_adxrs610JAVASCRIPT_wrap.cxx:1532:0:
/sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/git/src/adxrs610/./adxrs610.h:164:40: error: 'constexpr' needed for in-class initialization of static data member 'const float upm::ADXRS610::m_degreeCoeff' of non-integral type [-fpermissive]
     static const float m_degreeCoeff = 0.006;
                                        ^
/sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/git/src/adxrs610/./adxrs610.h:167:45: error: 'constexpr' needed for in-class initialization of static data member 'const float upm::ADXRS610::m_temperatureCoeff' of non-integral type [-fpermissive]
     static const float m_temperatureCoeff = 0.009;
                                             ^
/sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/git/src/adxrs610/./adxrs610.h:169:43: error: 'constexpr' needed for in-class initialization of static data member 'const float upm::ADXRS610::m_temperatureNom' of non-integral type [-fpermissive]
     static const float m_temperatureNom = 25.0;
                                           ^
src/adxrs610/CMakeFiles/jsupm_adxrs610.dir/build.make:67: recipe for target 'src/adxrs610/CMakeFiles/jsupm_adxrs610.dir/jsupm_adxrs610JAVASCRIPT_wrap.cxx.o' failed
make[2]: *** [src/adxrs610/CMakeFiles/jsupm_adxrs610.dir/jsupm_adxrs610JAVASCRIPT_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:11515: recipe for target 'src/adxrs610/CMakeFiles/jsupm_adxrs610.dir/all' failed
make[1]: *** [src/adxrs610/CMakeFiles/jsupm_adxrs610.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
WARNING: exit code 2 from a shell command.
ERROR: Function failed: do_compile (log file is located at /sde/r1.2.0/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+74691914fb-r0/temp/log.do_compile.19069)

BR,
xbolshe

xbolshe, 2015-12-13:
UPM has compiled successfully after disabling these 2 modules.

Propanu, 2015-12-15:
Hi, this happens because of the newly added C11 flags for JavaScript bindings. Interesting enough though, I can build all the jsupm modules just fine with GCC 4.8.2 and SWIG 3.0.5. There are several ways to approach this, from changing those variables to macros to enabling C11 for all the languages and switching to `constexpr`. I will take a look at the options.

xbolshe, 2015-12-29:
See https://github.com/intel-iot-devkit/upm/pull/326

xbolshe, 2016-01-07:
Same problem:

| /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/upm/0.4.1+gitAUTOINC+4ff722ed2e-r0/git/src/nlgpio16/./nlgpio16.h:86:35: error: 'constexpr' needed for in-class initialization of static data member 'const float upm::NLGPIO16::ADC_AREF' of non-integral type [-fpermissive]
|      static const float ADC_AREF = 3.3;
|                                    ^

Propanu, 2016-01-08:
I'm surprised those are the only ones causing issues, here's the full list:
```bash
-bash-4.2$ grep -rniI "const float"
otp538u/otp538u.cxx:147:  const float reference_vol= 0.5; // what is this value? (from seeedstudio)
otp538u/otp538u.cxx:148:  const float tempIncrement=10;
adxrs610/adxrs610.h:164:    static const float m_degreeCoeff = 0.006;
adxrs610/adxrs610.h:167:    static const float m_temperatureCoeff = 0.009;
adxrs610/adxrs610.h:169:    static const float m_temperatureNom = 25.0;
h3lis331dl/h3lis331dl.cxx:578:  const float gains = 0.003;    // Seeed magic number?
mg811/mg811.cxx:37:static const float dcGain = 8.5;
mg811/mg811.cxx:79:  static const float log400 = log10f(400);
mg811/mg811.cxx:80:  static const float log1000 = log10f(1000);
nlgpio16/nlgpio16.h:86:    static const float ADC_AREF = 3.3;
-bash-4.2$ grep -rniI "const double"                                                                                       
sx1276/sx1276.h:90:    static const double FXOSC_FREQ = 32000000.0;
sx1276/sx1276.h:94:    static const double FXOSC_STEP = 61.03515625;
```

I'm tempted to fix by changing to `constexpr` where names are lowercase and `#defines` where caps. Should work with GCC 4.5 or newer which I think is reasonable but I still have to check.

xbolshe, 2016-01-08:
Only constants like float and double defined in a namespace cause the issue.
No need to do with functions like log400 = log10f(400) or constants outside a namespace like cGain = 8.5.

Propanu, 2016-01-30:
The fix was merged, thanks a lot for providing it.

closed 2016-01-30 by Propanu