eclipse-upm/upm #325: jsupm_i2clcd npm module doesn't work on node.js nodejs 0.12.7 on Intel Edison

massca, 2015-12-22:
Hi,  I need to use node 0.12.x to run some 3rd party software that require this version of node.
I installed node 0.12.7 (from the Intel AlexT repo), and I was able to have mraa working (using its 0.8.0 version). I didn't find a workaround to have a working version of UPM jsupm_i2clcd module that I use to drive the LCD display.
This problem was reproduced by Intel_Peter on Edison forum, https://communities.intel.com/thread/96432

Thanks

Propanu, 2015-12-22:
Intel Galileo, Edison and Minnowboard MAX images are built with Node 0.10.36 and so are MRAA & UPM. However, you can build both libraries from source with a different (newer) version of Node. Version 0.12.7 was tested and builds correctly with no issues. For Node 5, you might have to make small modifications to the sources based on your compiler version due to C11 options. Also check #318.

massca, 2015-12-23:
Thanks for your answer!

deadprogram, 2016-01-21:
This is still an issue for anyone who is trying to use a newer version of Node.

Perhaps it would be good to leave this issue open?

deadprogram, 2016-01-21:
On Node 4.2.3 trying to `require("jsupm_i2clcd")` results in:

```
module.js:460
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: Module did not self-register.
    at Error (native)
    at Object.Module._extensions..node (module.js:460:18)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
...
```

alext-mkrs, 2016-01-21:
That's all the same - the error is caused by the fqct that the module is built for node 0.10.x, the default one in the image. For any other version  be it the one from my repo or self-built you'd neef to build upm by yourself. But that's really easy, so just follow the build instructions and you should be good.

deadprogram, 2016-01-21:
Thanks @alext-mkrs any tips on compiling for Node 4.2.3? I started with compiling mraa and got stuck here https://github.com/intel-iot-devkit/mraa/issues/411

alext-mkrs, 2016-01-22:
I haven't tried this specific combination, but I see @arfoll has already replied in the mraa issue you refer to, so let's continue there.

Propanu, 2016-03-04:
Will try to use node-gyp for UPM so that you can install the sensors from npm regardless of your installed node version, similar to what MRAA is doing. Leaving this open until that happens.

alext-mkrs, 2016-03-04:
Thumbs up, I think that would be a welcome enhancement

thiagogcm, 2016-04-02:
Looking forward to it!!!  @Propanu  :+1:

deadprogram, 2016-05-12:
Hi @Propanu any hope of publishing on npm a newer update of upm to match mraa 1.0 and also the latest drivers?

Propanu, 2016-05-12:
I don't have node-gyp working with UPM yet, but I can push a precompiled version like before. Will Node 4.4.4 do?

deadprogram, 2016-05-12:
That sounds great, thank you!

Propanu, 2016-05-14:
Publishing now:
* UPM 0.6.2 built with MRAA 0.10.1 & Node.js 0.10.33
* UPM 0.7.0 built with MRAA 1.0 & Node.js 4.4.4

Propanu, 2016-11-07:
As of UPM 1.0 source packages for the sensor modules are being pushed to NPM instead, giving the option to compile on any board against various MRAA/Node versions.

closed 2016-11-07 by Propanu