eclipse-upm/upm #327: Problem running st7735.js example for LCD

ShawnHymel, 2015-12-29:
I have an Adafruit 1.8" LCD based on the ST7735 driver. Whenever I try to run the [st7735.js](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/st7735.js) JavaScript example, I get the following error:                                                                                                                                            

```                                                                                                        
ERROR: /home/root/.node_app_slot/main.js:28                                                                                                                                 
ERROR: var lcd = new st7735.ST7735(10, 7, 9, 8);                                                                                                                            
          ^                                              
ERROR: Error: UPM Invalid Argument: Invalid GPIO pin specified                                                                                                              
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:28:11)                                                                                                         
    at Module._compile (module.js:456:26)                                                                                                                                   
    at Object.Module._extensions..js (module.js:474:10)                                                                                                                     
    at Module.load (module.js:356:32)                                                                                                                                       
    at Function.Module._load (module.js:312:12)                                                                                                                             
    at Function.Module.runMain (module.js:497:10)                                                                                                                           
    at startup (node.js:119:16)                                                                                                                                             
    at node.js:935:3
```

I've tried all different pin numbers, and nothing seems to change. I have a newly flashed Edison with the latest image, and MRAA is at v0.8.0. Any help would be appreciated.

Propanu, 2016-01-04:
Hi Shawn,

Does the C++ code work for you by any chance? This driver was definitely developed on a Galileo board and from my experience SPI drivers tend to be incompatible on the Edison.

For node, this might be caused by a missing `uint8_t` typemap in the binding. Also note that the CS, D/C and RESET pins are bit-banged manually in this implementation, so there should be no overlap with SPI pins (10-13). I have one of these displays on order and I'll take a look & update this once I get it.

Tudor

ShawnHymel, 2016-01-13:
The C++ example does not work. I get the following:

```
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Invalid GPIO pin specified
Aborted
```

When it is compiled and ran.

Propanu, 2016-04-11:
There was an issue with MRAA where the Edison Arduino board was not detected properly and instead the library would initialize for the mini-breakout. Either that or a damaged GPIO expander would usually cause this issue.

ShawnHymel, 2016-04-12:
Good to know, thanks! I've submitted the ili9341 module I was working on. Should we close this issue?

Propanu, 2016-04-12:
Yes, thanks again for the driver! By the way I saw your implementation and I think it could benefit a lot from using a buffer to send the data to the display similar to how it's done for the st7735 or ssd1351.

closed 2016-04-12 by Propanu

ShawnHymel, 2016-04-12:
Most welcome! I'm hoping to see more LCD solutions for the Edison and Galileo.

The buffer + refresh() method is definitely the preferred method, but what I found was that the SPI write() function on the Edison was EXTREMELY slow (it took over 20 seconds to refresh the screen). As a result, I had to perform writes as they were needed rather than refresh the whole screen. I've made mention of this to the Yocto team, and they said that the next release of the Edison image should have much faster SPI transfers. If that's the case, I'll look into updating the ILI9341 driver.

Propanu, 2016-04-12:
IoTDK 3.0 is now [available](https://software.intel.com/en-us/iot/hardware/edison/downloads). You'll find many new features and SPI works like a charm on the Edison from what I've seen.

ShawnHymel, 2016-04-18:
Awesome, thanks! I'll definitely check out the new image.