eclipse-upm/upm #329: Build error with Yocto 2.0

sakoman, 2016-01-06:
Error message:

ERROR: QA Issue: non debug package contains .debug directory: upm path work/core2-32-poky-linux/upm/0.4.0-r0/packages-split/upm/usr/lib/python3.4/site-packages/.debug/_pyupm_lm35.so [debug-files]
ERROR: QA run found fatal errors. Please consider fixing them.

Propanu, 2016-01-06:
Does this happen only for the LM35 module, or do you see it for all the python bindings?

sakoman, 2016-01-07:
I just get a single error on the LM35 module

xbolshe, 2016-01-07:
It looks like bitbake recipe bug, not upm package.

sakoman, 2016-01-08:
Ah, yes I filed the bug in the wrong place!

I use the meta-intel-iot-middleware layer (master branch).  Recipe is here:

http://git.yoctoproject.org/cgit/cgit.cgi/meta-intel-iot-middleware/tree/recipes-devtools/upm/upm_0.4.0.bb

Propanu, 2016-01-11:
Definitely a Yocto issue, just did a full install and I can't replicate it. Might want to sync with @arfoll on this as he wrote the bitbake recipes for the middleware libs.

closed 2016-01-11 by Propanu

alext-mkrs, 2016-01-12:
Based on my experience building upm for my package repo, the reason is most probably that the recipe was tested with the default Python 2, while in this case it's Python 3 it picks up. But the fix is anyway should be in the recipe, UPM rather can do nothing about this.

Propanu, 2016-01-12:
Yes, we do build with Python 2.7 instead of 3 to ensure compatibility with the IoTDK images.