eclipse-upm/upm #344: JHD1313M1 -- python how to pass number of column and rows

PranavManiar, 2016-02-08:
Hi I am using python to connecto to the groove backlight v4.0 LCD. 
It is 16x2 display, but the default seems to be something different. 

Is there a way to set the default column, row for the display ?? I searched through the documentation and could not find any way. 

Please suggest a way to display 16x2 text on display

alext-mkrs, 2016-02-08:
Actually jhd1313h1 is initialized as two-line one, you can see that by looking at constructor.

You need to set the cursor to the desired line and symbol and then write symbols out. See e.g. an example in my Edison workshop's code, edison-workshop repo, code/step3_lcd_ui/lcd_ui/lcd_ui.py

Propanu, 2016-02-10:
Hi,
I've modified the name of the node and python samples to match the driver name, jhd1313m1. You can find the python sample [here](../blob/master/examples/python/jhd1313m1-lcd.py). As @alext-mkrs mentioned, this LCD is initialized as a 2 row, 16 column display.

If you want to use a different configuration (say 1 x 32) you can use the LCM1602 driver directly which will allow you to specify the number of rows & columns. You will not have backlight controls, but you can still implement that manually using MRAA. Alternatively, you could do a bit of a hack and initialize both an LCM1602 for writing to the screen in a 1 by 32 format and an JHD1313M1 for controlling the backlight.

Propanu, 2016-05-03:
Closing as this has been answered.

closed 2016-05-03 by Propanu