eclipse-upm/upm #351: TH02 sensor visible [0x40 on i2c bus 6] but no measurement value transferred

g-vidal, 2016-02-19:
Hello,

I am using TH02 grove sensor connected to a grove board plugged on I2C connector of an Edison Arduino breakout. 

With `i2Cdetect -y -r 6` I can see that the sensor is recognized (0x40) but when I use the functions  `getHumidity()`  or `getTemperature ()` in my python script  I get the classical  -50 -24 values corresponding to 0 0 measured by the sensor.
A dump confirms that it is sending only 0 .
Trying the same sensor on an arduino gives correct values. I have compiles UPM and I'll try to investigate but I am not a dev and I am doing something wrong or thereis a bug in the python script .

Thanks for any help

g-vidal, 2016-02-21:
Hi,
I suspect a bug in TH02... Has anybody been able to get someting else than -50 -24 values ? 
Thanks.

Propanu, 2016-02-22:
Thanks for reporting the issue, I will update the list of known limitations once I confirm it!

g-vidal, 2016-02-22:
OK Perfect I stay tuned 
Thanks

HenkLaracker, 2016-03-02:
When i use the arduino ide and the groove indoor environment demo this sensor works fine, when i use eclipse (inteliot) with java and this library the TH02 sensor produces -50 -24 values. But ones in a while the correct temperature is shown. Any idea

g-vidal, 2016-03-02:
Sorry Henk, I have stopped wasting my time with that sensor... As  [Charles](uhttps://hallard.me/th02-library/rl) says in his post better to switch to another more reliable sensor.

> Tried to get it working on ULPNode with a TSL2861 (see this post), no way, this device claim as an I2C device, it’s just not, so I decided to switch to SI7021 sensor but this one seems now had it price raised twice on each reseller, no luck. !!!
Meanwhile your observation may suggest that the cpp library from "indoor environment demo" may have a trick to make this sensor work. Maybe it could be worth to compile ithis code inside upm and check what happens...If it works then pull a request to upgrade upm library. :-)

Good luck !

HenkLaracker, 2016-03-03:
Hi, i'm unfortunately not an expert in this area but i found out that they use : https://github.com/Seeed-Studio/Grove_Temper_Humidity_TH02

randomswdev, 2016-07-14:
If I switch Edison to 3V and set the switch on the Grove Shield to 3V, the sensor starts to work fine.
But the LCD backlight monitor stops working. Which is the correct configuration to have both of them working?

randomswdev, 2016-07-14:
Setting Edison to 3V and the Grove Shield to 5V appears to solve the issue both with the temperature sensor and the LCD display. Is it the correct configuration?

Propanu, 2016-07-14:
If you don't plan on adding any more sensors yes. However keep in mind that with the board at 3V3 and the shield at 5V you won't get the full range for the ADC, and adding another I2C sensor might make the bus unstable again.

Propanu, 2016-08-29:
Sensor is now listed on known limitations page as it is only usable at 3V3 on the Intel Edison with the Arduino board.

closed 2016-08-29 by Propanu