eclipse-upm/upm #352: pyupm_nrf24l01 source and destination address giving me issues.

aafeliz, 2016-02-19:
Hi, 
 I am currently working on creating a class that contains an NRF24L01 object. I keep getting an error stating that the addresses need to be uint8_t*, any ideas on how to get that done on python. I noticed that there were other sensors that used some sort of method that was define as part of the class. However pyupm_nrf24l01 does not contain such method. 
////////////////////////////////////////////////////////////////
import pyupm_nrf24l01 as trans

class Transceiver(object):
    def __init__(self, ch=99, srcAddress=[0x01, 0x01, 0x01, 0x01, 0x01], destAddress=[0x01, 0x01, 0x01, 0x01, 0x01]):
        self.module = trans.NRF24L01(45, 46)#7,8

        self.module.setSourceAddress(srcAddress)
        self.module.setDestAddress(destAddress)
        self.module.setPayload(trans.MAX_BUFFER)
        self.module.setChannel(ch)
        self.module.configure()
        self.module.setDataReceivedHandler(self.NRFHandler())
        self.module.setSpeedRate(trans.NRF_2MBPS)

    def NRFHandler(self):
        print("recieved data: " + str(self.module.m_rxBuffer[:]))

    def sendData(self, data):
        self.module.m_txBuffer = data
        self.module.send()

/////////////////////////////////////////////////////////////////
I tried using the following line: self.module.setSourceAddress(srcAddress). However it then gives me an error telling me that it is exception two parameters but I am trying to pass 6. I also noticed that there are no examples using pyupm_nrf24l01. I only noticed that it is implemented in c++(not Python).

Propanu, 2016-02-22:
Hi, the python & node.js interfaces for the NRF24L01 are incomplete and missing typemaps for pointers/arrays, thus they will have to be updated. So far this driver has only been tested in C++ on a Galileo Gen 2 board.

aafeliz, 2016-02-25:
do you have any suggestions for a temporary solution because I currently working on a robot project and time is of the essence.

Propanu, 2016-02-26:
The Intel Edison has integrated WiFi and BLE. For the Galileo boards we really recommend a mPCIe wireless card (e.g. Intel 6235). You can find out more about some of the supported cards in this [thread](https://communities.intel.com/thread/47847?start=0&tstart=0). The benefit of this solution is dual-band WiFi + BLE in one package.

malikabhi05, 2016-03-30:
Hi @aafeliz , the typemaps have been added for python and nodejs for nrf24l01, you should now be able to create a uint8Array using something like:  

y = pyupm_nrf24l01.uint8Array(5)

the array is initially populated with garbage values:

>>> y[0]
56L
>>> y[1]
24L

as of now I don't know of an easy way to populate the array all at once, however it seems to work individually:

>>> y[0] = 0x01
>>> y[0]
1L

and it can be passed to the function without receiving the error:

>>> x = pyupm_nrf24l01.NRF24L01(1,1)
>>> x.setDestinationAddress(y)
>>>

We haven't yet created any examples for python and nodejs for this sensor, however we will be doing so soon. Hopefully, the approach described above works.

Propanu, 2016-03-31:
@aafeliz please let us know if this works for you, or if you chose an alternative solution instead. Thanks!

Propanu, 2016-06-01:
Considering this resolved with the addition of the new typemaps.

closed 2016-06-01 by Propanu