eclipse-upm/upm #354: Cannot add sensor mq135 in upm lib. Linking CXX problem (?) looking for help.

g-vidal, 2016-02-23:
Hi all,

**1.** I have bought an MQ135 sensor (without reading the list of available sensors), I have found an mq135 code for  Arduino and I am trying to migrate it  in upm.... 
I have git cloned mraa and upm built and installed them. I have a first question because on the Edison  **Linking CXX shared library any.so
** is in red color while on my computer it is in bright green color while _Building CXX object src/sensor/CMakeFiles/_pyupm_sensor.dir/sensor.cxx.o_ is in both cases in dark green.

Is that a problem ? Am I missing a package on the Edison, is it necessary to install some packages before building mraa and upm ?

**2.** After the code migration I have cloned another  folder from upm/src named it mq135 then  changed all sensor names  to mq135 in the filenames and inside all files. I built this new module (I tried standalone and after clearing the build directory with the same result). Everything went well if I except the red color but it is the same for all modules. I was so proud ;-)....
Unfortunately when I try to import the module it founds it but it fails with "unknown reference" which usually means that there has been a linking problem.
Here is the message : 
```
>>> import pyupm_mq135
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/pyupm_mq135.py", line 28, in <module>
    _pyupm_mq135 = swig_import_helper()
  File "/usr/lib/python2.7/site-packages/pyupm_mq135.py", line 24, in swig_import_helper
    _mod = imp.load_module('_pyupm_mq135', fp, pathname, description)
ImportError: /usr/lib/python2.7/site-packages/_pyupm_mq135.so: undefined symbol: _ZN3upm5MQ135D1Ev
>>> 
```
I suspected that mathlib might not be linked and I tried target_link_libraries()  with no effect
Can anybody help me with this question please? mq135 are air quality captors of the Clickboard family and may interest several uers...

Thanks
Gérard

g-vidal, 2016-02-24:
Solution found  !
I used the  command line derived from : 
[http://stackoverflow.com/questions/5108079/how-do-i-find-out-which-functions-of-a-shared-object-are-used-by-a-program-or-an]( http://stackoverflow.com/questions/5108079/how-do-i-find-out-which-functions-of-a-shared-object-are-used-by-a-program-or-an)
in my case it becomes :
`target="src/mq135/libupm-mq135.so"; for symbol in $(nm -D $target | grep "U " | cut -b12-); do for library in $(ldd $target | cut -d ' ' -f3- | cut -d' ' -f1); do for lib_symbol in $(nm -D $library | grep "T " | cut -b12-); do if [ $symbol == $lib_symbol ]; then echo -e "Found symbol: \e[1;36m$symbol\033[0m at \e[1;34m$library\033[0m"; fi ; done; done; done;`

Which enables to find which reference is unknown (I had declared but not defined ~MQ135)
If you suffer a problem like mine you may also find usefull those commands found in other places :
```
readelf -Ws src/mq135/libupm-mq135.so
nm -gC src/mq135/libupm-mq135.so
objdump -TC src/mq135/libupm-mq135.so

```
I'll check the MQ135 program and propose it when it will work well.
Hope this may help some of you.

Propanu, 2016-02-24:
Hi there!
Regarding your questions:
1. You don't have to worry about the colors not showing up. That's a feature added by cmake and in some shells it works while others won't display them at all. For instance with default settings on Putty I never get any sort of colored output from my connections ever.
2. This is a very odd scenario, never had linking problems before as long as the right libraries were referenced. May I ask what math library are you using, anything beyond glibc "math.h"? For that you don't need any additional linker flags. Otherwise `target_link_libraries()` as you mentioned should do the trick. For example you can look at the [CMakeLists.txt](../blob/master/src/stepmotor/CMakeLists.txt) file for the stepmotor driver which was recently revamped and needed to link against librt for older versions of GCC.

g-vidal, 2016-02-25:
Hi Propanu,
On question 1  I was just wondering why it was green on one machine and green on another, usually when you receive red messages among a green flow you have to worry ;-) that's why I asked. 
On question 2 I had seen it and checked it as I wrote in my first post. To clarify what happened I had declared in the mq135.h a closer ~MQ135 but unfortunately forgot to define it in mq135.cxx. The funny thing is that it passes the compiling step without detecting the undefined function and the detection happens with linking which confused me. I am testing  the captor I'll make a pull request next week if things remain functionning.

Thanks again

closed 2016-04-13 by Propanu