eclipse-upm/upm #356: TSL2561 sensor python example problem

dmiller100, 2016-02-24:
Hello, I have been trying to use the UPM library for the tsl2561 with python 2.7 on an intel Edison.
When I run tsl2561.py I get the following error;

Traceback (most recent call last):
  File "tsl2561.py", line 28, in <module>
    myDigitalLightSensor = upmTsl2561.TSL2561()
  File "/usr/lib/python2.7/site-packages/pyupm_tsl2561.py", line 263, in __init__
    this = _pyupm_tsl2561.new_TSL2561(*args)
RuntimeError: UPM Runtime Error: TSL2561: Unable to power up TSL2561

The sensor works fine using a modified adafruit library.

All I can see wrong is that in tsl2561.h
#define TSL2561_Address          (0x29)  //Device address
should be
#define TSL2561_Address          (0x39)  //Device address

I have tried changing this but has no affect, not sure all what has to be done.

Any help appreciated
Cheers

Propanu, 2016-02-24:
Hi! The address for the Grove Digital Light sensor from SeeedStudio is 0x29 and that is our default, but you have the option of passing a different address to the constructor like so:

`myDigitalLightSensor = upmTsl2561.TSL2561(0, 0x39)`

Modifying the header file won't do anything unless you recompile the sources. Our documentation page for the sensor has more details, you can find it for python [here](http://iotdk.intel.com/docs/master/upm/python/pyupm_tsl2561.html). Also, please note that this is one of the sensors listed in our list of [known limitations](../blob/master/docs/knownlimitations.md).

dmiller100, 2016-02-25:
Thanks, all working fine now.

closed 2016-02-25 by dmiller100