eclipse-upm/upm #357: ADS1115 and javascript

Kimlorentz, 2016-02-25:
How do I  write commands to ads1115 and read data?

How to set:
*SPS = 32 
*PGA = 2/3
*DIFF0_1

After setup how do I ask for DIFF0_1 data

Propanu, 2016-02-26:
The base ADS1x15 class provides all the functionality you need. You can set the sample rate with the `setSPS()` function and the gain with `setGain()`. In your case you would use:
```
setSPS(32)
setGain(GAIN_TWOTHIRDS)
```
After setting it up you can get the data with `getSample()` which returns DIFF_0_1 data by default. For further details please refere to the javascript API page for the [ADS1x15 module](http://iotdk.intel.com/docs/master/upm/node/modules/ads1x15.html).

Kimlorentz, 2016-02-26:
I tryed this code.

```
var mraa = require('mraa');
var version = mraa.getVersion();

if (version >= 'v0.6.1') {
    console.log('mraa version (' + version + ') ok');
}
else {
    console.log('mraa version(' + version + ') is old - this code may not work');
}

// Instantiate on I2C bus
var ADS1X15 = require('jsupm_ads1x15');
var ads1115 = new ADS1X15.ADS1115(1, 0x48); 
ads1115.setSPS(32);
ads1115.setGain(GAIN_TWOTHIRDS);

setInterval(function(){
    
    console.log("*********************");
    console.log("Sample: " + ads1115.getSample()); // Read DIFF 0-1
    console.log("Gain: " + ads1115.getGain());
    console.log("SPS: " + ads1115.getSPS());
    console.log("*********************");
    
}, 1000); 
```

Only got this error:

```
Upload Complete 
mraa version (v0.9.4) ok  
  
ERROR: /home/root/.node_app_slot/main.js:15 
ERROR: ads1115.setGain(GAIN_TWOTHIRDS);
    ERROR:             ^
ERROR: ReferenceError: GAIN_TWOTHIRDS is not defined
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:15:17)    
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10) 
    at startup (node.js:119:16)
    at node.js:935:3
```

yannickadam, 2016-02-26:
I don't have this device so I can't really try it. 
However, GAIN_TWOTHIRDS is not defined on the global scope. 
You probably need to use something like ADS1X15.GAIN_TWOTHIRDS

Propanu, 2016-02-26:
Yeah, the enum is nested in the class, try `ADS1X15.ADS115.GAIN_TWOTHIRDS`, and for SPS it's probably `ADS1X15.ADS115.SPS_32`.

Propanu, 2016-06-01:
Considered solved.

closed 2016-06-01 by Propanu