eclipse-upm/upm #359: Cannot use 2 sensors on I2c bus

g-vidal, 2016-02-27:
Hi all
This issue is driving me crazy ! I cannot plug 2 sensors on I2c Edison arduino breakout.
I am using :
[Altitude Click mpl3115a2](http://www.mikroe.com/click/altitude/) whose datasheet is  [here](http://cache.nxp.com/files/sensors/doc/data_sheet/MPL3115A2.pdf?pspll=1) and 

[Adafruit Temp  and humidity HTU21D-F](https://www.adafruit.com/products/1899) whose datasheet is [here](https://www.adafruit.com/datasheets/1899_HTU21D.pdf).
When plugging Adafruit only I get this :
```python
>>> import pyupm_htu21d
>>> bus = 6
>>> tempAddress = 0x40
>>> temp = pyupm_htu21d.HTU21D(bus, tempAddress)
>>> temp.resetSensor ()
>>> temp.sampleData()
0
>>> RH = temp.getCompRH()
>>> temperature = temp.getTemperature()
>>> hygro = temp.getHumidity ()
>>> print '\t%.3f\t\t'%temperature + '%.3f\t\t'%hygro + '%.3f'%RH
	22.477		33.550		33.928
```
When plugging  Altitude only I get that :
```python
>>> import pyupm_mpl3115a2
>>> bus = 6
>>> pressAdress = 0x60
>>> press = pyupm_mpl3115a2.MPL3115A2(bus, pressAdress)
>>> press.sampleData()
0
>>> temperatureP = press.getTemperature()
>>> pressure = press.getPressure()
>>> pressea = press.getSealevelPressure()
>>> alt = press.getAltitude()
>>> print '\t%.3f\t\t'%temperatureP + '%.3f\t\t'%pressure + '%.3f\t\t'%pressea + '%.3f'%alt
	23.062		97761.750		97761.750		300.980
```
Given my altitude to 180 m and the pressure 997 hP We can consider those measurements as correct with the need of calibration.
If I connect both captors at the same time, both become crazy and  forward the following measurements or failures :
```python
>>> temp.resetSensor ()
>>> temp.sampleData()
0
>>> RH = temp.getCompRH()
>>> temperature = temp.getTemperature()
>>> hygro = temp.getHumidity ()
>>> print '\t%.3f\t\t'%temperature + '%.3f\t\t'%hygro + '%.3f'%RH
	-46.850		31.109		41.886
>>> 
>>> press.sampleData()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/pyupm_mpl3115a2.py", line 223, in sampleData
    return _pyupm_mpl3115a2.MPL3115A2_sampleData(self)
RuntimeError: UPM Runtime Error: i2cWriteReg:mraa_i2c_write() failed
>>> temperatureP = press.getTemperature()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/pyupm_mpl3115a2.py", line 277, in getTemperature
    return _pyupm_mpl3115a2.MPL3115A2_getTemperature(self, bSampleData)
RuntimeError: UPM Runtime Error: i2cWriteReg:mraa_i2c_write() failed
>>> pressure = press.getPressure()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/pyupm_mpl3115a2.py", line 261, in getPressure
    return _pyupm_mpl3115a2.MPL3115A2_getPressure(self, bSampleData)
RuntimeError: UPM Runtime Error: i2cWriteReg:mraa_i2c_write() failed
>>> pressea = press.getSealevelPressure()
>>> alt = press.getAltitude()
>>> print '\t%.3f\t\t'%temperatureP + '%.3f\t\t'%pressure + '%.3f\t\t'%pressea + '%.3f'%alt
	23.062		97761.750		97761.750		300.980
```
I have checked many tricks, changing the order of the captors, adding pull-up resistors changed the sensors, use GROVE shield with no effect..
Has anybody else observed such a behavior or succeded in connecting more than one sensor ?
Is it possible to solve this bug?

rwaldron, 2016-02-29:
This may or may not be helpful but the MPL3115A2 is the only I2C sensor that I've encountered which requires sending a "restart" after read or write: 

- https://github.com/adafruit/Adafruit_MPL3115A2_Library/blob/master/Adafruit_MPL3115A2.cpp#L170-L184
- https://www.arduino.cc/en/Reference/WireEndTransmission

Propanu, 2016-02-29:
There are a few sensors that require a repeated start after sending a command, and unfortunately I don't think MRAA gives this option as of now. In fact, there are a few sensors claiming to be I2C compliant and actually they're not, leaving bit-banging as the only option to get a response from them.

Won't hurt to open a request for this feature on the MRAA github page!

Also, if you plan on using several I2C sensors in your project I highly recommend getting a Mini-breakout or the Sparkfun blocks for your Edison as you get access to both I2C buses and they are a lot more stable.

g-vidal, 2016-02-29:
Hi,
Thanks rwaldon for the information, it must be the problem. Do you know any pressure/temperature altitude  sensor that would not require a "repeated start", we are trying to build "IoT-weatherstations for education" and we need the more fluent solutions to guarantee success for the children....

Thanks also Propanu sorry to bother you  with things that look like evidences but I am new to this world... I have started a discussion on  INTEL forum [here](uhttps://communities.intel.com/message/376533#376533rl)  with no answer yet, I'll read MRAA GitHub issues and see if the topic is out  and contribute. 
To answer your question I do have a mini breakout :-) I'll try to use it but  that will not solve my MPL3115A2 problem. I'll let you know how it works I did not use it because I am also using Gas captors that need higher voltage and more power to heat the sensor, I thought it would have been simpler with arduino breakout.
It should be very usefull for newcomers like me to have a page describing with some details the limitations (telling for example not fully I2c compliant) and maybe ask on such a page for "best practices"  : which sensor is yielding good or best results in a given situation ?
I have not found the limitations page you cited in other messages .

Many thanks again to you both.

Propanu, 2016-02-29:
Try this [link](../blob/master/docs/knownlimitations.md) for the limitations page.

whbruce, 2016-07-25:
Suggest rewording title of issue to indicate this is specific to Edison hardware.

Propanu, 2016-08-31:
Added MPL3115A2 to the list of known limitations. HTU21D patch was merged too, we can continue the discussion in the separate thread #457

closed 2016-08-31 by Propanu