eclipse-upm/upm #370: Intel edison problem using sensor TSL2561

hannes93, 2016-03-15:
Hey there,
I like to use the Intel Edison and the TSL2561 Light lux sensor.
the down loaded lib dosen't work.
Is ther some body out who has the same problem and is there any solution for that problem?

Thank you very much for your help!!!

Greetz \h

malikabhi05, 2016-03-15:
Hi @hannes93 , could you please give us some more details on what you're trying to do and where you face the issue in order to understand the problem better, also which language you are using? Please note that this sensor is listed in the [known limitations](https://github.com/intel-iot-devkit/upm/blob/master/docs/knownlimitations.md) page as well.

Propanu, 2016-06-01:
Unfortunately, as @malikabhi05 mentioned this sensor is known to be incompatible with the Intel Edison Arduino breakout. Closing as inactive/no additional information provided.

closed 2016-06-01 by Propanu