eclipse-upm/upm #372: Use upm with nodejs version 4.4.0 on Intel Edison

AlessandroH, 2016-03-17:
Could someone help me?Hi,
I need to use nodejs 4.4.0 on Intel Edison and I tried to compile upm starting with "make jsupm_i2clcd"  with many errors. The first one:
make jsupm_i2clcd
[100%] Building CXX object src/lcd/CMakeFiles/jsupm_i2clcd.dir/jsupm_i2clcdJAVASCRIPT_wrap.cxx.o
In file included from /usr/include/node/node.h:42:0,
                 from /home/root/upm/build/src/lcd/jsupm_i2clcdJAVASCRIPT_wrap.cxx:156:
/usr/include/node/v8.h:336:1: error: expected unqualified-id before 'using'
 using Handle = Local<T>;
  Is there a version 4.4.0 compatible? Could someone help me?
Thanks, Alessandro

Propanu, 2016-03-17:
Hi, the drivers should build with 4.1.2 and 5, @xbolshe added the changes required for that. We haven't tested against nodejs 4.4.0 but this error indicates the V8 version string is off and likely does not enable C11 support. You can force it with `-DCMAKE_CXX_FLAGS:STRING=-std=c++11` when you configure if you're only building this sensor (it will fail on full builds until I merge #364). I recommend using `ccmake` to generate the makefiles since you can see all the configuration variables/paths in advanced mode.

Let me know if that works for you!

AlessandroH, 2016-03-18:
Hi, THANKS .. it work's!!!!!!!!!!!!

closed 2016-03-18 by AlessandroH