eclipse-upm/upm #376: Problem with installation path

xbolshe, 2016-03-26:
After https://github.com/intel-iot-devkit/upm/commit/a48f457057748d190e9faeec6495ef9a8c504de6 NODE_MODULE_INSTALL_PATH is related with NODE_ROOT_DIR. And it creates a problem with changed sysroot dir and cross-compiled nodejs.

Here is a part of install_manifest.txt:

/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/node_modules/jsupm_grovevdiv/package.json
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/node_modules/jsupm_grovevdiv/jsupm_grovevdiv.node
/usr/lib/libupm-grovevdiv.so.0.6.0
/usr/lib/libupm-grovevdiv.so.0
/usr/lib/libupm-grovevdiv.so
/usr/include/upm/grovevdiv.h
/usr/lib/pkgconfig/upm-lpd8806.pc
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/node_modules/jsupm_lpd8806/package.json
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/node_modules/jsupm_lpd8806/jsupm_lpd8806.node
/usr/lib/libupm-lpd8806.so.0.6.0
/usr/lib/libupm-lpd8806.so.0
/usr/lib/libupm-lpd8806.so
/usr/include/upm/lpd8806.h
/usr/lib/pkgconfig/upm-groveeldriver.pc
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/node_modules/jsupm_groveeldriver/package.json
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/node_modules/jsupm_groveeldriver/jsupm_groveeldriver.node
/usr/lib/libupm-groveeldriver.so.0.6.0
/usr/lib/libupm-groveeldriver.so.0
/usr/lib/libupm-groveeldriver.so
/usr/include/upm/groveeldriver.h
/usr/lib/pkgconfig/upm-grovescam.pc
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/node_modules/jsupm_grovescam/package.json
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/node_modules/jsupm_grovescam/jsupm_grovescam.node
/usr/lib/libupm-grovescam.so.0.6.0
/usr/lib/libupm-grovescam.so.0
/usr/lib/libupm-grovescam.so
/usr/include/upm/grovescam.h
/usr/lib/pkgconfig/upm-flex.pc

As you may see NODE_ROOT_DIR is added to modules paths.

Now the full path is {SYSROOTDIR}/{NODE_ROOT_DIR}/lib/node_modules/*/*.node.

xbolshe, 2016-03-26:
I have checked my environment. Here what I have in macro(upm_SWIG_NODE):

NODE_MODULE_INSTALL_PATH = /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark
/usr/lib/any/

CMAKE_CURRENT_BINARY_DIR = /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky
-linux/upm/0.6.0+gitAUTOINC+230bc4f569-r0/git/build/src/any

PROJECT_SOURCE_DIR = /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux
/upm/0.6.0+gitAUTOINC+230bc4f569-r0/git

NODE_ROOT_DIR = /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr

CMAKE_SYSROOT = <empty>

xbolshe, 2016-03-26:
If I return DESTINATION lib/node_modules/jsupm_${libname} back, the problem goes away.

Propanu, 2016-03-28:
I'm guessing something similar to https://github.com/intel-iot-devkit/mraa/commit/14de175425d0e315e5a130cdd9d0c6c8b7026cce will fix this issue.

Propanu, 2016-03-29:
Please let me know if 02974f242efd59aaf1bc54e98796cd8c8c43310a fixes this for you.

xbolshe, 2016-03-29:
It does not fix the problem:

usr/lib/node_modules/jsupm_grovegprs/package.json
/sdc/ww25.5-15/edison-src/out/linux64/build/tmp/sysroots/edison/usr/lib/node_modules/jsupm_grovegprs/jsupm_grovegprs.node
/usr/lib/libupm-grovegprs.so.0.6.0
/usr/lib/libupm-grovegprs.so.0
/usr/lib/libupm-grovegprs.so
/usr/include/upm/grovegprs.h
/usr/lib/pkgconfig/upm-lm35.pc
/usr/lib/node_modules/jsupm_lm35/package.json
/sdc/ww25.5-15/edison-src/out/linux64/build/tmp/sysroots/edison/usr/lib/node_modules/jsupm_lm35/jsupm_lm35.node
/usr/lib/libupm-lm35.so.0.6.0
/usr/lib/libupm-lm35.so.0
/usr/lib/libupm-lm35.so
/usr/include/upm/lm35.h
/usr/lib/pkgconfig/upm-groveeldriver.pc
/usr/lib/node_modules/jsupm_groveeldriver/package.json
/sdc/ww25.5-15/edison-src/out/linux64/build/tmp/sysroots/edison/usr/lib/node_modules/jsupm_groveeldriver/jsupm_groveeldriver.node
/usr/lib/libupm-groveeldriver.so.0.6.0
/usr/lib/libupm-groveeldriver.so.0
/usr/lib/libupm-groveeldriver.so
/usr/include/upm/groveeldriver.h
y-linux/upm/0.6.0+gitAUTOINC+02974f242e-r0/git/build$

xbolshe, 2016-03-29:
Fix for package.json, but not for any.node

Propanu, 2016-03-30:
Missed a `set` command that was still using node_root. Should be fixed for both now with db8d7087988e846c28a3a7ae61cf0d82905a154c. Thanks!

xbolshe, 2016-03-30:
Yes, now it works correctly.

Propanu, 2016-03-31:
Thanks for the confirmation, closed!

closed 2016-03-31 by Propanu