eclipse-upm/upm #377: include "upm/iTemperatureSensor.h"

xbolshe, 2016-03-26:
There is an error:

| Scanning dependencies of target jsupm_cjq4435
| [ 35%] Building CXX object src/cjq4435/CMakeFiles/jsupm_cjq4435.dir/jsupm_cjq4435JAVASCRIPT_wrap.cxx.o
| [ 35%] Building CXX object src/cjq4435/CMakeFiles/jsupm_cjq4435.dir/cjq4435.cxx.o
| [ 35%] Linking CXX shared module jsupm_cjq4435.node
| [ 35%] Built target jsupm_cjq4435
| Scanning dependencies of target bmpx8x
| [ 35%] Building CXX object src/bmpx8x/CMakeFiles/bmpx8x.dir/bmpx8x.cxx.o
| In file included from /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/upm/0.6.0+gitAUTOINC+230bc4f569-r0/git/src/bmpx8x/bmpx8x.cxx:31:0:
| /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/upm/0.6.0+gitAUTOINC+230bc4f569-r0/git/src/bmpx8x/bmpx8x.h:33:36: fatal error: upm/iTemperatureSensor.h: No such file or directory
|  #include "upm/iTemperatureSensor.h"
|                                     ^
| compilation terminated.
| src/bmpx8x/CMakeFiles/bmpx8x.dir/build.make:62: recipe for target 'src/bmpx8x/CMakeFiles/bmpx8x.dir/bmpx8x.cxx.o' failed
| make[2]: *** [src/bmpx8x/CMakeFiles/bmpx8x.dir/bmpx8x.cxx.o] Error 1
| CMakeFiles/Makefile2:5846: recipe for target 'src/bmpx8x/CMakeFiles/bmpx8x.dir/all' failed
| make[1]: *** [src/bmpx8x/CMakeFiles/bmpx8x.dir/all] Error 2

It seems that the following code is missed in macro(upm_SWIG_NODE):

    include_directories (
      ${CMAKE_CURRENT_SOURCE_DIR}/..
    )

g-vidal, 2016-03-26:
I have got the same problem  and something in the same vein for t6713....
Has anybody got a woraround?
this works for me
cp upm/iCO2Sensor.h  t6713/
cp upm/iModuleStatus.h t6713/
cp upm/iTemperatureSensor.h bmpx8x
cp upm/iModuleStatus.h bmpx8x
cp upm/iADC.h ads1x15
cp upm/iModuleStatus.h ads1x15

Thanks

xbolshe, 2016-03-26:
See https://github.com/intel-iot-devkit/upm/pull/378

Propanu, 2016-03-29:
Merged #378 but this looks like a C++ build failure. We'll likely have to update the CMakeLists.txt file with the include per sensor.

xbolshe, 2016-03-29:
What was a reason to add ${CMAKE_CURRENT_SOURCE_DIR}/.. to macro(upm_SWIG_JAVA) ? I guess macro(upm_SWIG_NODE) has the same reason.

xbolshe, 2016-03-29:
Also it is related with a building directory. I use <root_upm>/git/build to build with autotools.

Propanu, 2016-03-30:
Parent directory was now added to all the targets and I expect this approach to work regardless of build directory.

Propanu, 2016-03-31:
Thanks for the feedback, closed!

closed 2016-03-31 by Propanu