eclipse-upm/upm #379: Cannot compile upm properly anymore, cannot get new modules

g-vidal, 2016-03-27:
Hi I have got a strange problem

I did a git pull from intel-iot-devkit/upm  on my Edison and when I try to compile I get errors  like this :
` Nothing known about base class 'ITemperatureSensor'. Ignored.`
The python modules that succeed arrive there :
`/usr/lib/python./site-packages/`
instead of  :
`/usr/lib/python2.7/site-packages/`
I did not change anything in the process
```shell
cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr ..
make
sudo make install
```
I cannot figure out what happens for the BMP180 that I need and more generally for the modules that do not appear :
```python
>>> import pyupm_bmp180
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named pyupm_bmp180
```
Any help would be greatly appreciated, I have prepared a training session and nothing works any more ;-(
Thanks

xbolshe, 2016-03-28:
It seems that ${PYTHON_VERSION_MAJOR} and ${PYTHON_VERSION_MINOR} are not detected.

g-vidal, 2016-03-29:
Follow-up....

I deadly need to be able to compile mraa and ump libs with students on various platforms : Edison / RaspBerry-Pi  PClinux Debian, then I have set up a workaround:

- As suggested by @xbolshe   I wrote fixed values for  ${PYTHON_VERSION_MAJOR} and ${PYTHON_VERSION_MINOR}  in upm/src/CMakeLists.txt 
- I removed all declarations like public ILightSensor, public upm::ILightController, public ITemperatureSensor, public IHumiditySensor, public IPressureSensor that introduce warnings and sometimes compilation errors
- I think that for multiple reasons including science teacher responsability we need to spell properly scientists names then I wrote properly Anders Celsius name included in some functions.

I have not had time to check if the functions were properly working but they compile and at least they can be imported from python prompt ! If anybody is interested the workaround is there : [on branch upm+GV](https://github.com/g-vidal/upm.git).
If anybody with better skills than me can bring back upm lib in the original fluent compilation mood it will be greatly appreciated ;-).

Propanu, 2016-03-29:
Python issues should be fixed now with e8090161522962fff3ef38ea8445f9cfac49cd24. Many new features were introduced with version 0.6.0, and unfortunately we still need to ensure compatibility with all the possible build combinations. When using this in a sensitive environment (academia or production) I highly recommend sticking to a stable version (v.0.5.1) until the new release is reinforced in case you encounter build errors. Thanks a lot for your feedback and patience!

g-vidal, 2016-03-29:
Many many thanks for your work !

>
> Python issues should be fixed now with e809016
> Many new features were introduced with version 0.6.0, and
> unfortunately we still need to ensure compatibility with all the
> possible build combinations. When using this in a sensitive
> environment (academia or production) I highly recommend sticking to a
> stable version (v.0.5.1) until the new release is reinforced in case
> you encounter build errors.
yes I know :-[ :-[  You are right  I should  butO:-) ..... It is  so fun
to ride the wave....8-)
>
> Thanks a lot for your feedback and patience!
With pleasure, in my opinion the important point is to teach how to make
it work, in this cas eunfortunately the problem was at the very
beginning of the process... I did not notice that we jumped from 0.5 to
0.6....
I will check again and let you know.

Best regards

Propanu, 2016-04-11:
Closing, as v0.6.1 should fix all these issues. Thanks again!

closed 2016-04-11 by Propanu