eclipse-upm/upm #381: JavaScript Example for MAX31855

jmwright, 2016-03-31:
I see UPM code for the MAX31855 and was wondering if there are any examples of reading the MAX31855 from JavaScript. The import seems to work fine on an Edison, but I was hoping to get some guidance on how to set it up to read correctly. Even an example of something that's somewhat close to the MAX31855 in operation would be helpful.

https://github.com/intel-iot-devkit/upm/blob/master/src/max31855/max31855.cxx

jmwright, 2016-03-31:
The following JavaScript code doesn't throw any errors, but always returns 0 for the temperature. I've checked the device using SPI on an Arduino Uno and it works fine. I could certainly have a wiring issue on the Edison, but I've checked it many times and can't find where the problem might be.
```javascript
var max31855 = require('jsupm_max31855');
var tc = new max31855.MAX31855(0, 31);
var temp = tc.getTemp();
console.log("Temp: " + temp);
```

jmwright, 2016-03-31:
If I ground the CS pin manually I can get it to work (reads a correct temperature). I see that the jsupm_max31855 library is writing the pin to low, so there must be an issue with my hardware setup.

closed 2016-03-31 by jmwright

Propanu, 2016-03-31:
Hi, you can use the table on this [page](http://iotdk.intel.com/docs/master/mraa/edison.html) to find MRAA/UPM -> physical pin mappings for the Mini-breakout. Pin 31 in code would be J19-4. You could use MRAA's GPIO class to toggle the pin and test it separately. On the Arduino breakout only pins 0 through 19 are exposed. For custom boards you can also use the raw pin names as defined [here](https://github.com/intel-iot-devkit/mraa/blob/master/api/mraa/types.h#L112).