eclipse-upm/upm #384: can't hx711 sensor with nodejs

bymafmaf, 2016-04-05:
Hi!

I've installed mraa and can confirm it's working. I tried many things and lastly just installed jsupm_hx711 with:
`npm install jsupm_hx711`

But I'm getting this error: 
```
module.js:355
  Module._extensions[extension](this, filename);
                               ^

Error: /home/pi/Desktop/nodejs/main/node_modules/jsupm_hx711/jsupm_hx711.node: cannot open shared object file: No such file or directory
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/home/pi/Desktop/nodejs/main/hx711.js:2:13)
    at Module._compile (module.js:434:26)
    at Object.Module._extensions..js (module.js:452:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)

```

Working on Raspberry Pi 2 Model B.

Propanu, 2016-04-05:
Hi, UPM npm modules currently include prebuilt binaries for i586 and node 0.10.39, thus they won't work on anything else than an Intel board running that version of node. We plan to change that in the future but unfortunately as of now your only option is to compile the hx711 module from source on the RPi2. The [building](../blob/master/docs/building.md) document has more details.

bymafmaf, 2016-04-05:
I tried that way before realizing there was a npm module but couldn't figure out what to use as node module. Also I want to give a feedback that nodejs documentation is poor, especially for software guys.

Propanu, 2016-04-05:
You just need to install [Node.js](https://nodejs.org/en/download/) (includes binaries + headers) for your platform and that should enable you to build the js bindings. Cmake will find the right paths automatically so in most cases nothing else needs to be set.

bymafmaf, 2016-04-06:
Now I get `Segmentation Fault` error which I think related to my nodejs setup. Completely removed it. What do you mean by includes binaries and headers? Is downloading arm7 version and `sudo make install` enough?

Propanu, 2016-04-06:
You need the node header files to be able to build our libraries, but they are included in the Node.js distribution I've mentioned (as opposed to, for instance, installing node with nvm AFAIK). If you download the arm7 version, you can simply extract the tarball and move the contents of the folder to `/usr` for instance and cmake will find this when building MRAA & UPM.

Propanu, 2016-08-29:
Considered resolved. Feel free to reopen if you are still having issues building our libraries on RPi.

closed 2016-08-29 by Propanu